iNavFlight/inav #2485: Disabling stick command profile selection

joeclone, 2017-11-17:
I'm running 1.7.3 on a flying wing. Crashed the other day coz accidentally changed profiles while on ground fiddling with the tx sticks. Don't see any mention in the wiki about possibility of disabling profile selection by stick command. Would like to have the option to do that like how blheli does, disabling tx programming.

digitalentity, 2017-11-17:
Seems like a nice feature to me. I also only use one profile and accidentally switching to another one may cause bad stuff to happen.

We can dedicate a feature bit to enabling profile selection (enabled by default).

joeclone, 2017-11-17:
Thats good to hear. Yea enabled at default and maybe a tick box on pid page option to disable by the user.  Thanks.

Redshifft, 2017-12-13:
Excellent idea + this would enable us to use 2 stick arming with less fear of "in Air" disarm and save another switch or two (if you run an arm guard switch)

digitalentity, 2018-02-25:
#2830 merged. Closing

closed 2018-02-25 by digitalentity