iNavFlight/inav #2487: Mixed up Rx options

happyuser007, 2017-11-17:
### Board and Version
NAZE32 full
INAV 1.7.3

# version
# INAV/NAZE 1.7.3 Sep  9 2017 / 09:59:03 (974440df)

### Behavior
It seems that there is a mix up of options for Rx mode in the "Configuration" tab Zone "Receiver mode". I have wired my receiver to have a ppm output but I cannot get the channels in the "Receiver" tab if I choose "PPM Rx input". I get all channels with my single input if I choose PWM input (one wire per channel). Is this a bug in the configurator software? Could you check and correct that behavior if it is true because it is weird especially for noobies like me.
Thank you for maintaining that forum.

fiam, 2017-11-18:
This is a bug in the configurator, it's fixed in this PR (not yet merged) https://github.com/iNavFlight/inav-configurator/pull/303

fiam, 2018-02-26:
Fixed in configurator

closed 2018-02-26 by fiam