iNavFlight/inav #2490: Failsafe procedure SET-THR not taking on Failsafe Throttle value if ALT HOLD enabled.

KiwiDavid, 2017-11-20:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
* _Steps needed to reproduce the problem_
* _Expected Results_
* _Actual Results_

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.

_Upload a zipped blackbox log (if available)_

digitalentity, 2017-11-21:
Duplicate of https://github.com/iNavFlight/inav/issues/1872
We somehow didn't merge the fix for the bug https://github.com/iNavFlight/inav/pull/1873

teckel12, 2017-11-21:
@digitalentity This issue did spark an idea.  If the failsafe mode is LAND/SET-THR, wouldn't it be a good idea to turn on POS HOLD (assuming GPS and a lock obviously)?  At 400' on a windy day, my quad could easily end up in the river where I typically fly from.  I would much rather it come directly down than drift who knows how far.

I think this would also be helpful as telemetry would show the last GPS coordinates, which would be very close to where the craft would land, instead of basically anywhere if the throttle is just reduced and it's allowed to drift.

I use RTH for failsafe, but those using LAND and have GPS I would think would rather know where their craft is instead of it just drifting to the unknown.

KiwiDavid, 2017-11-22:
That would seem like a good idea and from the perspective of public safety a predictable, vertically descending craft is probably preferable to one drifting in the breeze.
I could also use RTH but wonder what happens if you don't have the battery capacity to make it back and hence I prefer to be on the ground quickly.
Similarly I would like to see the option of a low battery trigger for Failsafe SET-THR- something the craft's FC can decide all on its own and not dependent on operator, transmission link or even receiver function.

darmach, 2017-11-22:
@teckel12 @digitalentity It would be great if the failsafe landing would POSHOLD when possible (when GPS lock is established). How about making the failsafe landing more "intelligent"? The old behaviour - sticking to predefined throttle for predefined time - is a little bit dull, isn't it? Could it be made to utilize altitude information for an intelligent landing?

teckel12, 2017-11-22:
@darmach Exactly. This is INAV afterall, not BF.

teckel12, 2017-11-22:
I would think LAND would work basically like the final part of RTH, where it would descend at a set speed with POSHOLD. Obviously, the landing altitude wouldn't be unknown so it couldn't change descent speed as RTH does. Once we work in the code that knows if the craft is in the air or not, it could also disable the motors on landing like RTH.

Leaving SET-THR as-is seems like good backwards compatibility as well. I see SET-THR and LAND as two different options, each which make sense and can both exist.

darmach, 2018-03-20:
How's it going with this one? Maybe at least the bug with alt hold failsafe could be fixed?

digitalentity, 2018-03-21:
@darmach @KiwiDavid please re-test with 1.9

KiwiDavid, 2018-03-21:
Firmware version 1.9. Hardware SPRacing F3 clone. Test platform quad 250. Failsafe procedure is SET-THR (formerly LAND). Set Throttle value 1340
Test condition - POS HOLD and ALT HOLD then force failsafe with tx OFF.
With quad hovering at about 1.5m it behaves as expected and does fast, controlled descent to ground then does motor shut down after about 20 sec.  Happy with this.   Will test without Pos Hold later when wind is low and I have bigger space available.

KiwiDavid, 2018-03-22:
Sorry I hit the wrong button and while the issue looks like it's fixed it still needs a little more testing to confirm.

KiwiDavid, 2018-03-22:
Just confirming that the Set-THR/Land is also working as expected without Pos Hold engaged.
Same test as above but without Pos Hold used.  Looking good.

Cheers,
David

KiwiDavid, 2018-03-26:
Original issue is now corrected in 1.9 and with no further input from others I think this can now be closed

closed 2018-03-26 by KiwiDavid