iNavFlight/inav #2493: Wrong mode after return from failsafe/RTH procedure

Allerhopp, 2017-11-20:
iNav 1.8

After exiting the failsafe/RTH with stick movement the prior modes are cancelled. Only "armed" is applied. No blackbox log, but smartport telemetry Temp1 shows the modes:

Before failsafe: 12215, in failsafe 52215, after failsafe 10005

1 = flaperon mode, 2 = auto tune mode, 4 = failsafe mode
1 = return to home, 2 = waypoint mode, 4 = headfree mode
1 = heading hold, 2 = altitude hold, 4 = position hold
1 = angle mode, 2 = horizon mode, 4 = passthru mode
1 = ok to arm, 2 = arming is prevented, 4 = armed

teckel12, 2017-11-20:
@Allerhopp Are you saying that Tmp1 = 52215 during FAILSAFE but when you exit FAILSAFE Tmp1 = 10005?  You seem to report conflicting information in: https://github.com/iNavFlight/LuaTelemetry/issues/54

teckel12, 2017-11-20:
@Allerhopp Are you sure you're running INAV 1.8.0?  Tmp1 of 10005 would mean that FLAPERON mode was still enabled.  Seems strange that that would be the only mode activated after FAILSAFE was disabled.  

As there were changes in INAV 1.2.0 and LuaTelemetry 1.8.0, I want to be sure that you're running the latest of each as older versions may show different results.  Also, did you have FLAPERON enabled?

Allerhopp, 2017-11-21:
Btw. the mode text messages are copied from the iNav code, they are for wings, but i fly a copter.

Yes, running INAV 1.2.0 and LuaTelemetry 1.8.0.
Hmmm, i do not see conflicting information in iNavFlight/LuaTelemetry#54. These are two differnet issues.

Nearly lost my copter, when this happened the fist time, because i did not excpect to end in Acro after exiting the failsafe procedure. I saved the copter entering RTH manually.
[X4R_fs.zip](https://github.com/iNavFlight/inav/files/1490116/X4R_fs.zip)

Allerhopp, 2017-11-21:
It seems, that iNav activates itself altitude hold and angle mode when the user activates waypoint mode. After a failsafe-exit iNav cancels not only waypoint (this is, what i expected) but also altitude hold and angle mode. In my OpenTX config altitude hold and angle mode are not activated automatically together with waypoint. This explains eventually, why iNav falls back to acro mode.

I presume, that after a failsafe-exit iNav shall cancel waypoint mode and it´s self activated altitude hold and angle mode. Is this the expected behaviour? In this case the user has to take care, which flightmode is active after failsafe-exit of waypoint mode and i have to rework my flightmode selection.

digitalentity, 2017-11-21:
This is intended behavior. WP mode is self-contained, it will activate ANGLE + ALTHOLD automatically. However, if WP will fail to activate machine may end up in unexpected state. 

I believe we should amend the documentation and advise users to activate ANGLE mode manually for safety.

teckel12, 2017-11-21:
This makes sense. But what still doesn't is that Tmp1 of 10005 would mean FLAPERON was enabled if running 1.8.0. With 1.8.0, I would expect just a value of 5 instead. Either a mistype or you're running 1.7.3 or earlier

Allerhopp, 2017-11-21:
# version
# INAV/SPRACINGF3 1.8.0 Oct 24 2017 / 06:57:49 (0f1975ce)

I started WAYPOINT from FLAPERON/ACRO mode. If i understand it correctly, the fallback after failsafe will be FLAPERON/ACRO again, 10005 seems to make sense.
This happens now, when failsafe is finished by stick movement: ANGLE is activated, because my waypoint selection selects now ANGLE also. Then i activated WAYPOINT manually again.
I am very happy with the result. Please feel free to close this issue.
[image]

teckel12, 2017-11-21:
I guess I was confused because I thought you said you were flying a quad but you have flaps? This led me to believe you may have been running a different version as this just changed in 1.8.0.

closed 2018-05-27 by stale[bot]