iNavFlight/inav #2494: Mag HMC5883 on Revo not detected

pipersw, 2017-11-22:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
_Version of INAV used goes here_
configurator 1.8.1 and firmware 1.8

_Use CLI command `version` and put its output here_

_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_

Since upgrade from configurator 1.7.3 and firmware 1.7.3 to configurator 1.8 and firmware 1.8, the magnetometer HMC5883 on the CC3D REVO is not detected.
Same Behavior with configurator 1.8 and firmware 1.7.3

* _Steps needed to reproduce the problem_

Upgrade the configurator to 1.8.1, goes to configuration tab and the field for the mag is set to NONE.
log: 
2017-11-22 @ 08:21:12 -- Running - OS: Windows, Chrome: 62.0.3202.94, Configurator: 1.8.1
2017-11-22 @ 08:22:33 -- Serial port successfully opened with ID: 1
2017-11-22 @ 08:22:33 -- MultiWii API version received - 2.1.0
2017-11-22 @ 08:22:33 -- Flight controller info, identifier: INAV, version: 1.8.0
2017-11-22 @ 08:22:33 -- Running firmware released on: Nov 1 2017 06:39:22
2017-11-22 @ 08:22:33 -- Board: REVO, version: 0
2017-11-22 @ 08:22:33 -- Unique device ID received - 0x3d00563535511031313931

If I select HMC5883, save and reboot, the configurator close the serial port after I want to connect it:

2017-11-22 @ 08:23:26 -- EEPROM saved
2017-11-22 @ 08:23:26 -- Device - Rebooting
2017-11-22 @ 08:23:27 -- Serial port successfully closed

then I do a reconnection and then the field show always NONE, and the configurator want an upgrade of the firmware whereas it's already the last revision:

2017-11-22 @ 08:23:30 -- Serial port successfully opened with ID: 2
2017-11-22 @ 08:23:31 -- MultiWii API version received - 2.1.0
2017-11-22 @ 08:23:31 -- Flight controller info, identifier: INAV, version: 1.8.0
2017-11-22 @ 08:23:31 -- Running firmware released on: Nov 1 2017 06:39:22
2017-11-22 @ 08:23:31 -- Board: REVO, version: 0
2017-11-22 @ 08:23:31 -- Unique device ID received - 0x3d00563535511031313931
2017-11-22 @ 08:23:32 -- Serial port successfully closed
2017-11-22 @ 08:23:33 -- You need to upgrade your firmware before you can use the Configuration tab.

* _Expected Results_

Mag HMC5883 detected as before

* _Actual Results_

mag HMC5883 not detected

digitalentity, 2017-11-22:
Are you referring to built-in mag or external mag?

pipersw, 2017-11-22:
it's a built-in mag, and I don't have an external mag connected to the board.

digitalentity, 2017-11-22:
This is a known issue and intentional change to allow using external compass.
We'll allow chosing between internal and external compass in 1.9

pipersw, 2017-11-27:
So, You mean that you could choice between I2C1 bus for internal HMC5883 or I2C2 bus for external HMC5883 without to have to remove the internal HMC5883 ?

digitalentity, 2017-11-27:
@pipersw yes, exactly. Not tomorrow, but in 1.9 we'll have this option on some boards

pipersw, 2017-11-29:
I have just connect an external HMC5883 on Flexiport and the compass is seen when I choose HMC5883.

digitalentity, 2017-11-30:
Yes, this is how it works in 1.8

PR #2522 (2017-12-04): Enable selection between compass sensors on different buses
New option `mag_to_use` will allow to select betwen compass sensor on internal/external I2C bus (on boards where built-in compass is on separate I2C bus - only REVO so far).
Fixes #2494
closed 2018-02-12 by digitalentity