iNavFlight/inav #2496: Omnibus F4 V3 clone Tricopter Motor 3 output not linear

SpamsterFly, 2017-11-22:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.
I have verified with multiple FC, ESC is verified with other firmware.  Triflight and betaflight do not exhibit this problem with the same configuration.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
Betaflight Omnibus F4 Pro V3 Flight Controller
https://us.banggood.com/Wholesale-Warehouse-30_5x30_5mm-Betaflight-Omnibus-F4-Pro-V3-Flight-Controller-Built-in-OSD-Barometer-SD-Blackbox-wp-Usa-1163085.html?rmmds=myorder

_Version of INAV used goes here_
1.8
_Use CLI command `version` and put its output here_
INAV/OMNIBUSF4V3 1.8.0 Nov  1 2017 / 06:35:50 (912d1315)
Also tried Pro firmware, same results
_If it's a custom-compiled firmware please mention this_

### Behavior
* _Describe the problem_
Using motors tab in configurator, motor 3 doesn't spin up until about half throttle, then response is non-linear, decreases at full throttle and is 'jumpy'
* _Steps needed to reproduce the problem_
Setup V3 clone as tricopter with motors connected pwm1-3 and servo pwm 6
Spin motor on motors tab
* _Expected Results_
motor 3 should respond like 1 and 2
* _Actual Results_
Motor 3 is unresponsive until ~50% throttle, then is not smooth or linear response

_Is this a new feature request?_

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.
# dump

# version
# INAV/OMNIBUSF4V3 1.8.0 Nov  1 2017 / 06:35:50 (912d1315)

# resources

# mixer
mixer TRI

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 0 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
[... 512 lines cut ...]
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25
set tpa_rate = 20
set tpa_breakpoint = 1650

# 
_Upload a zipped blackbox log (if available)_

shellixyz, 2017-11-26:
Can you reproduce with standard PWM at 50Hz update ?

SpamsterFly, 2017-11-27:
I just reflashed and set to tricopter, set motor to PWM @ 50 Hz and the behavior is the same.

digitalentity, 2017-11-27:
Can't reproduce. Motor 3 responds to sliders on "Motors" tab correctly. Tested PWM, OneShot125 and Multishot. Re-calibrate your ESC.

DzikuVx, 2017-12-12:
There seems to be a stall over here, so closing. In case new info appears, feel free to reopen

closed 2017-12-12 by DzikuVx