iNavFlight/inav #2498: Trying to get softserial to work on OmnibusF4 PRO

shabbirun, 2017-11-23:
I've got an Omnibus F4 PRO clone which I've set up iNav 1.8(latest) on.

I am trying to get SoftSerial to work, but there are no CH5 and CH6 pads on the board so I tried to remap the pins for PWM5 and PWM6 in the source code to Softserial RX and TX.

target.c shows Motor 5 and Motor 6 as PA1 and PA8.

`#if (defined(OMNIBUSF4PRO) || defined(OMNIBUSF4V3)) && !defined(OMNIBUSF4PRO_LEDSTRIPM5)
    { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // MOTOR_5
    { TIM4,  IO_TAG(PB6),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM4, TIM_USE_LED                                            }, // LED strip for F4 V2 / F4-Pro-0X and later
#else
    { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // MOTOR_5
#endif
    { TIM1,  IO_TAG(PA8),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // MOTOR_6`

I changed the Softserial mapping in target.h to PA1 and PA8.

`#if defined(OMNIBUSF4V3)
#define SERIAL_PORT_COUNT       4 //VCP, USART1, USART3, USART6
#else
#define USE_SOFTSERIAL1
#define SOFTSERIAL_1_RX_PIN     PA1
#define SOFTSERIAL_1_TX_PIN     PA8
`

After hooking up the bluetooth module and enabling softserial, there's still no communication between the board and the computer/phone. Baudrate is 9600 to match the module.

Any ideas?

shellixyz, 2017-11-24:
It's been some time since I worked on this board but I think CH5 and CH6 have pads on the board. The difference is that these pads are not drilled through. They are just plain pads like that https://static.rcgroups.net/forums/attachments/5/9/3/2/6/3/a9583299-16-55BFCE96-B634-4F0F-AB7C-3059B4E33B53.jpg

teckel12, 2017-11-29:
@shabbirun Are you sure you don't have the small pads as shown here?:

https://github.com/iNavFlight/inav/blob/development/docs/assets/images/omnibusf4pro_ss.jpg

closed 2018-05-27 by stale[bot]