iNavFlight/inav #2499: Last motolab version is 1.3

jcaamnes, 2017-11-23:
Hi
is it possible to get the latest version of inav on a motolab board? I only able to flash version 1.3.
[image]

might be related to [https://github.com/iNavFlight/inav/issues/1757](url)

jcaamnes, 2017-11-23:
i was able to build motolab target using this [guide](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Mac%20OS%20X.md)

jcaamnes, 2017-11-23:
When i change mixer to fixed wing. the resources is lost.

How can i assign different pins to resources?
Fixed wing:
A00: BEEPER OUT
A01: FREE 
A02: FREE 
A03: FREE 
A04: FREE 
A05: FREE 
A06: FREE 
A07: FREE 
A08: FREE 
A09: I2C2 SCL
A10: I2C2 SDA
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: MPU EXTI
B00: FREE 
B01: FREE 
B02: FREE 
B03: FREE 
B04: FREE 
B05: LED1 OUT
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX
B08: FREE 
B09: LED2 OUT
B10: FREE 
B11: FREE 
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE 

Quad setup
A00: BEEPER OUT
A01: MOTOR5 OUT
A02: MOTOR6 OUT
A03: MOTOR7 OUT
A04: MOTOR1 OUT
A05: FREE 
A06: MOTOR2 OUT
A07: FREE 
A08: MOTOR8 OUT
A09: I2C2 SCL
A10: I2C2 SDA
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: MPU EXTI
B00: MOTOR3 OUT
B01: MOTOR4 OUT
B02: FREE 
B03: FREE 
B04: FREE 
B05: LED1 OUT
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX
B08: FREE 
B09: LED2 OUT
B10: FREE 
B11: FREE 
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE

shellixyz, 2017-11-24:
Fixed wing support has not been enabled for this FC yet.

You can try this patch if you feel confident enough: https://github.com/shellixyz/inav/tree/MOTOLAB_FW_support

The only difference between this and stock 1.8 is that I enabled motors on PWM 1-2 and servos on PWM 3-6 for fixed wing mixers.

jcaamnes, 2017-11-24:
Thank you for your help. i build and flashed target for motolab using your code. Now motor 1 works with fixed wing mixer, but servos dont. I dont think servos are mapped

resources looks like this now:
A00: BEEPER OUT
A01: MOTOR5 OUT
A02: MOTOR6 OUT
A03: FREE 
A04: MOTOR1 OUT
A05: FREE 
A06: MOTOR2 OUT
A07: PPM IN
A08: FREE 
A09: I2C2 SCL
A10: I2C2 SDA
A11: FREE 
A12: FREE 
A13: FREE 
A14: FREE 
A15: MPU EXTI
B00: MOTOR3 OUT
B01: MOTOR4 OUT
B02: FREE 
B03: FREE 
B04: FREE 
B05: LED1 OUT
B06: SERIAL1 UART TX
B07: SERIAL1 UART RX
B08: FREE 
B09: LED2 OUT
B10: FREE 
B11: FREE 
B12: FLASH CS
B13: SPI2 SCK
B14: SPI2 MISO
B15: SPI2 MOSI
C13: FREE 
C14: FREE 
C15: FREE 
F00: FREE 
F01: FREE 
F03: FREE 
F04: FREE

digitalentity, 2017-11-24:
@jcaamnes please try on a live board. Both motors and servos are mapped as "MOTOR" in resources.
@shellixyz can you submit a PR for FW support in INAV?

jcaamnes, 2017-11-24:
In servo tab I enabled live mode. Then tried to connect servos to motorpin 2-8.
On motorpin 3 and 4 i se movement on servo 3 and 4. No of the other pins worked.

Next step will be to remap servo 3 and 4 to 1 and 2 i guess?. that should be possible with smix?

shellixyz, 2017-11-24:
@digitalentity Sure it is now PR #2502 
@jcaamnes I don't understand exactly what you tested. On what type of model are you using this FC ? No signal on outputs 7-8 is normal. For flying wings the elevons should be connected to output 3 and 4. For airplanes elevator on 3, ailerons on 4-5 and rudder on 6. On the configuration tab/page next to the mixer selection you should see a connection diagram with the outputs to use for each surface corresponding to the selected mixer.

jcaamnes, 2017-11-25:
I thought motor pin 3 and 4 shared the same timer as 1 and 2 therefore my understanding was that i had to use motor pin 5-8. Sorry for the missunderstanding. when i arm now, everything seems to work (just bench testing now). thank you so much for your help, I really appreciate it 👍

digitalentity, 2017-11-25:
@jcaamnes @shellixyz indeed. pins 1-4 use the same timer. As long as you configure motor to conventional PWM and 50Hz update rate - you're fine, but we need to add a safety check for this condition in #2502

shellixyz, 2017-11-27:
@jcaamnes Please could you try out this branch:
https://github.com/shellixyz/inav/commits/MOTOLAB_FW_support_complete

jcaamnes, 2017-11-29:
Sorry for late reply. i will try your branch and report back as soon as possible.

jcaamnes, 2017-12-04:
Hi again 
I have been busy lately with work and family, but finally i have got time to test the new target.
unfortunately, I dont get it to work properly. 
When i change values on motor 1 in inav configurator the servos are moving, the motor dont respond. i have tried to connect the servos to different pins, to see if it works with no luck.

To be sure how should i wire the different components? i have motor on PWM 1.
tried servos on PWM 3-4, 5-6, 7-8. What is the correct wiring?

[image]

jcaamnes, 2017-12-04:
I reflashed back to inav-MOTOLAB_FW_support. here everything seems to work fine. i have motor ong PWM1 and servos on PWM 3-4. 

motor 1 is using oneshot125, while servos use PWM 50hz.

shellixyz, 2017-12-07:
@jcaamnes Good news, so it is working without the timers change. Are you sure you built the right branch because your build log mentions the master branch ? Anyway it is not a problem I compiled it for you: [inav_1.8.0_MOTOLAB.hex](https://drive.google.com/open?id=1cIkIdoM3JvsopTKB0y3Z_HcMLUJFWtaW). Please test this version it is better than the MOTOLAB_FW_support branch because with the old branch even though you selected oneshot125 for the motor it is outputing standard 50Hz PWM. It should work because I tested the modification on another target using the same MCU and output pins .

jcaamnes, 2017-12-11:
i cannot get it to work. i flashed the hex file you sendt me @shellixyz.
i connected motor to PWM1, servo to PWM3 and 4. set esc protocol to standard. When i arm, the motor is working, but i dont see any movements on the servos.

shellixyz, 2017-12-12:
@jcaamnes Indeed, my fault. It should be fixed: [inav_1.8.0_MOTOLAB.hex](https://drive.google.com/open?id=1sfi9ll_FGXIzVpEi71PqSbk1I4RBJAOk)

jcaamnes, 2017-12-12:
perfect 👍  now it works. thanks alot @shellixyz

shellixyz, 2017-12-12:
Great :)

The fix has already been merged and should be available for the next version. Can we close this issue ?

closed 2017-12-13 by jcaamnes