iNavFlight/inav #2503: busdev confused by inverted power on for SPRF3

stronnag, 2017-11-24:
SPRacing F3 Acro (genuine) and Dodo (SPRF3 firmware)

`# INAV/SPRACINGF3 1.8.1 Nov 23 2017 / 17:15:33 (39e89fc4)`

Following on from https://github.com/iNavFlight/inav/pull/2290#issuecomment-346190323 I confidently expected by Dodo vtail to arm today. Again, it didn't; CLI reason "ANGLE" (yet it's sitting level on the park bench).

On investigation, this happens if I power the machine inverted (something which has worked for ever). Now, with busdev and SPRF3, the machine things it's inverted even when it's level.

Having powered the machine inverted, three beeps for gyro OK, I turn it so it's the right way up.

[image]

Here's what the configurator thinks
[image]

This does not happen on e.g. an Omnibus, I can power on inverted and it's OK when normally level.

stronnag, 2017-11-24:
diffs in case it matters (same used for both pre/post busdev firmware)
````
# mwptools / cf-cli dump 2017-11-23T19:06:12+0000
# dumping condensed 'diff all' settings
# cf-cli is a tool to manage Cleanflight CLI dump backup and restore
# <https://github.com/stronnag/mwptools>

# version
# INAV/SPRACINGF3 1.8.1 Nov 23 2017 / 17:15:33 (39e89fc4)

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mixer VTAIL4

# servo

# servo mix

# feature
feature -BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map
map TAER5678

# name
name vtail

# serial
serial 1 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 2 2 900 2100

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 8
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6050
set acczero_x = 64
set acczero_y = 34
set acczero_z = 110
set accgain_x = 4064
set accgain_y = 4087
set accgain_z = 4063
set mag_hardware = NONE
set baro_hardware = NONE
set pitot_hardware = NONE
set receiver_type = SERIAL
set rssi_channel = 9
set serialrx_provider = SPEK2048
set rx_min_usec = 985
set blackbox_rate_denom = 32
set min_throttle = 1060
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_off_delay = 50
set failsafe_throttle = 1200
set vbat_scale = 112
set vbat_warning_cell_voltage = 34

# profile
profile 1

set mc_p_pitch = 55
set mc_i_pitch = 40
set mc_d_pitch = 15
set mc_p_roll = 55
set mc_i_roll = 40
set mc_d_roll = 15
set mc_p_yaw = 90
set mc_d_yaw = 20
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25
set tpa_rate = 20
set tpa_breakpoint = 1650

# profile
profile 2

# profile
profile 3

# restore original profile selection
profile 1

# save configuration
````
and

```` mwptools / cf-cli dump 2017-11-24T21:06:58+0000
# dumping condensed 'diff all' settings
# cf-cli is a tool to manage Cleanflight CLI dump backup and restore
# <https://github.com/stronnag/mwptools>

# version
# INAV/SPRACINGF3 1.8.1 Nov 19 2017 / 12:13:31 (1833bfd3)

# reset configuration to default settings
defaults noreboot

# resources

# mixer
mixer TRI

# servo
smix reverse 5 2 r

# servo mix

# feature
feature -BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper
beeper -ON_USB

# map
map TAER5678

# name

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 2 0 900 2100

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 8
set gyro_hardware_lpf = 256HZ
set gyro_lpf_hz = 90
set gyro_notch1_hz = 400
set gyro_notch1_cutoff = 300
set gyro_notch2_hz = 200
set gyro_notch2_cutoff = 100
set acc_hardware = MPU6050
set acczero_x = 24
set acczero_y = 9
set acczero_z = -125
set accgain_x = 4059
set accgain_y = 4076
set accgain_z = 4092
set mag_hardware = NONE
set baro_hardware = NONE
set pitot_hardware = NONE
set receiver_type = SERIAL
set serialrx_provider = SPEK2048
set rx_min_usec = 985
set min_throttle = 1080
set max_throttle = 2000
set motor_pwm_rate = 2000
set motor_pwm_protocol = MULTISHOT
set failsafe_off_delay = 50
set failsafe_throttle = 1350
set vbat_warning_cell_voltage = 33
set servo_pwm_rate = 330
set servo_lpf_hz = 160
set tri_unarmed_servo = OFF
set 3d_deadband_throttle = 1000
set nav_extra_arming_safety = OFF
set async_mode = GYRO

# profile
profile 1

set mc_p_pitch = 55
set mc_i_pitch = 40
set mc_d_pitch = 15
set mc_p_roll = 55
set mc_i_roll = 40
set mc_d_roll = 15
set mc_p_yaw = 90
set mc_d_yaw = 20
set dterm_lpf_hz = 80
set dterm_notch_hz = 260
set dterm_notch_cutoff = 160
set roll_rate = 70
set pitch_rate = 55
set yaw_rate = 25
set tpa_rate = 20
set tpa_breakpoint = 1650

# profile
profile 2

# profile
profile 3

# restore original profile selection
profile 1

# save configuration
````

digitalentity, 2017-11-24:
Can you try recalibrating accelerometer and making a test (and a diff)
again?

# <https://github.com/stronnag/mwptools>

# <https://github.com/stronnag/mwptools>

or mute
the thread

stronnag, 2017-11-24:
Sure it will be easier to dismantle the dodo machine I think (I have a special 6 point 'jig').

stronnag, 2017-11-24:
Very different accgains (all other settings are the same). 
````
set acczero_x = 50
set acczero_y = 41
set acczero_z = 49
set accgain_x = 2032
set accgain_y = 2041
set accgain_z = 2041
````
Drift of c. 1 deg/sec in when just sitting 'stationary' has gone, power on inverted shows inverted with install display of level once the board is returned to normal attitude.

digitalentity, 2017-11-25:
Ok, this is definitely related to MPU6050 revision detection.

digitalentity, 2017-11-28:
@stronnag can you please test the #2519? Re-calibrate and check that accgain values are as usual?

stronnag, 2017-11-28:
I'll test this (UTC) evening.

digitalentity, 2017-11-28:
Thanks. I managed to reproduce the issue, found and fixed a stupid return type bug which was missed by the compiler.

stronnag, 2017-11-28:
Tested on the DODO board. The mega-drift has gone and the calibration looks sensible again.

**New values**
````
set acczero_x = 47
set acczero_y = 41
set acczero_z = 18
set accgain_x = 4065
set accgain_y = 4083
set accgain_z = 4081
````
**Pre-busdev** 
````
set acczero_x = 64
set acczero_y = 34
set acczero_z = 110
set accgain_x = 4064
set accgain_y = 4087
set accgain_z = 4063
````

stronnag, 2017-11-28:
However, it I merge development + de_mpu6050_fix, then I see in the CLI & configurator CAL / "Run-time calibration" arm-check failures and the 3D-model can flip  on the small movements. Only tested on SPRF3 so far.

stronnag, 2017-11-28:
And while we're on the subject of (in this case harmless) compiler warnings:
````
./src/main/drivers/bus_busdev_i2c.c: In function 'i2cBusWriteBuffer':
./src/main/drivers/bus_busdev_i2c.c:31:89: warning: passing argument 5 of 'i2cWriteBuffer' discards 'const' qualifier from pointer target type [-Wdiscarded-qualifiers]
     return i2cWriteBuffer(dev->busdev.i2c.i2cBus, dev->busdev.i2c.address, reg, length, data);
                                                                                         ^~~~
In file included from ./src/main/drivers/bus.h:33:0,
                 from ./src/main/drivers/bus_busdev_i2c.c:26:
./src/main/drivers/bus_i2c.h:67:5: note: expected 'uint8_t * {aka unsigned char *}' but argument is of type 'const uint8_t * {aka const unsigned char *}'
 bool i2cWriteBuffer(I2CDevice device, uint8_t addr_, uint8_t reg_, uint8_t len_, uint8_t *data);
     ^~~~~~~~~~~~~~
````

digitalentity, 2017-11-28:
Ok, so the MPU revision detection issue is fixed. Merging and closing.

closed 2017-11-28 by digitalentity