iNavFlight/inav #2504: ALthold PID Tunning

DAquad, 2017-11-24:
### Board and Version
Omnibus F3 v1.1

_Use CLI command `version` and put its output here_INAV/OMNIBUS 1.7.3 Sep  9 2017 / 10:09:29 (974440df)

### Behavior
* When applying ALThold on my Tricopter (I use the 600mm Tricopter presents) I get unexcepted drop, I tried to follow the ALT+VEL PID Tuning steps from the GitHub navigation modes but can't find the set "nav_alt_p" command in CLI. Is it change?

My CLI dump

# version
# INAV/OMNIBUS 1.7.3 Sep  9 2017 / 10:09:29 (974440df)

# resources

# mixer
mixer TRI

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 -100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature GPS
feature CURRENT_METER
feature BLACKBOX
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
[... 469 lines cut ...]
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 35
set thr_mid = 50
set thr_expo = 0
set roll_rate = 55
set pitch_rate = 48
set yaw_rate = 20
set tpa_rate = 20
set tpa_breakpoint = 1650

Thanks for the help

#

shellixyz, 2017-11-24:
Config variable names have changed I think what you are looking for are these (from your dump):
set nav_mc_pos_z_p = 50
set nav_mc_pos_z_i = 0
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10

DAquad, 2017-11-24:
@shellixyz , Ohhhh I see....those are exist in the PID tuning tab in the configurator GUI as well.  
Cheers mate ☻

closed 2017-11-24 by DAquad