iNavFlight/inav #2505: Omnibus F3 Servo Outputs not correct on flying wing profile.

wolfcry0, 2017-11-25:
### Board and Version
Omnibus F3
1.8
INAV/OMNIBUS 1.8.0 Nov  1 2017 / 06:14:28 (912d1315)

### Behavior
On the flying wing mixer type it seems to be using Servos 4/5 instead of Servos 1/2 like the documentation and helper image shows. I used the "600mm flying wing" preset.

* _Expected Results_
Wing servos operate on PWM 3/4, with motors on 1/2.

* _Actual Results_
Wing servos are not on PWM 3/4.

digitalentity, 2017-11-25:
Tested on actual hardware. Wing servos are PWM3/4 as expected.

wolfcry0, 2017-11-25:
Very odd, this is a screenshot of the configurator with me pulling down on the elevator.

[image]

digitalentity, 2017-11-25:
Please note that both motor and servo graphs shown in Configurator reflect *logical* motor/servo number. Actual mapping to board PWM pins is board-dependent. @Painless360 did a great video explainging this feature: https://youtu.be/qL2LMFF5bJ4?t=2m31s

wolfcry0, 2017-11-25:
Well after doing some more testing it appears my board may be damaged and not providing any power to the PWM output headers +5V pin, so I'll close this issue.

closed 2017-11-25 by wolfcry0

digitalentity, 2017-11-25:
You board is probably fine. OMNIBUS boards doesn't provide power to PWM 5V rail. You need to use adedicated BEC.

wolfcry0, 2017-11-25:
Ah, well that does make sense. Weird that the docs I found for the F3 label them as +5V when they really just mean V+ or something.