iNavFlight/inav #2506: FW v1.8.0 / MatekF405OSD - PosHold+AltHold, RTH waving (with video)

Pairan, 2017-11-25:
### Board and Version
* INAV/MATEKF405OSD 1.8.0 Nov  1 2017 / 06:51:22 (912d1315)

### Behavior
* While getting into PosHold + AltHold or RTH the FW will start to go in waves up and down!
* This is a permanent effect at the moment in these flight modes

[I've a video ready that show the effect on (YouTube)](https://youtu.be/JFk8dY3a8lA)

### How to get it adjusted?
* I assume an configuration error here (which the must be my fault)
* What can I do? Has anybody had this?

shellixyz, 2017-11-26:
You should try to lower nav_fw_pos_z_p. I think a good value to try would be 30.

Maybe the default is a bit high ? I also had to lower it on my fixed wing builds.

Schalonsus, 2017-11-26:
This seems to be normal behaviour since iNav 1.7
With versions before this was not happening.
I played with every parameter possible and could not really solve this issue....played with nav_fw_pos_z_p and _i ..... then with level_p, level_i, level_d, with manual_climb_speed... auto_climb_speed..... no improvement at all. pitch2thr also did nothing positive
maybe some baro/gps conflict, dont know, never solved this problem. Only when going back to 1.6

shellixyz, 2017-11-26:
ALTHOLD PID loop parameters are nav_fw_pos_z_*. In theory if you reduce nav_fw_pos_z_p you should get rid of the oscillations. But a bad CG or thrust angle could also be a problem and it is possible you would need to lower nav_fw_pos_z_p too much for ALTHOLD to be effective. Depending on what device you are using for altitude estimation (GPS and/or baro) inav_gps_delay, inav_w_z_gps_p and  inav_w_z_baro_p also has an effect.

Pairan, 2017-11-27:
Altitude is provided by GPS, I will try to reduce the mentioned values. 
The funny thing about this:
I fly the exact same components on a S800 and there ist no trouble there in V.1.8.0 - lets see, if we can figure whats goind on. I will try to play with that on the rare flyable days and report back

Pairan, 2017-11-27:
actually, there is another odd thing here - the buzzer is constantly on in some situations! You can hear it in the video. Thought I might had misconfigured something here but the modes are fine and turn the buzzer on if chan 8 is set to max. However it starts and stops by itself. 

Maybe I reflash the Matek and do a full chip erase as these issues are'nt "normal" :)

shellixyz, 2017-11-27:
Different model different tuning it is not so surprising.

About the buzzer issue are you sure it is the buzzer and not the sound from the switching of your ESC ? Can you reproduce the problem on the ground ? Doing a full chip erase probably won't do anything but can always try. If you really have a buzzer issue you should report it separately.

Pairan, 2017-11-27:
Pretty sure ... it's the same sequence as if you trigger it via switch (MODES). Inititally I thought it would be my action cam ... but the footage delivers proof that it's not the action cam (= otherwise there would be no video :D)
By now I wasnt able to reproduce it on the ground. Maybe I should take the prop off, power up the motor and try to trigger the effect! It doesnt happen all the time ... but in certain moments (Usually im in horizon mode, auto tune often on). 

Putting it on the to do list! and yes, it'll become a separte issue then.

Pairan, 2018-01-02:
**Waving / overshooting /  oszilating**
I'm still on this topic ... in the user group a bunch of others with an AR WING seem to get similar effect. I wonder what is so special on that wing that several people get this effect there!

Schalonsus, 2018-01-02:
Maybe you could try a much older firmware like 1.3
I think this problem came with the latest versions.
I also have this problem on several planes that did not have the problem before. Just after upgrading iNav the pumping started.

I also realized that version 1.8 is much less appreciated for racing... Downgraded my race wing to 1.7 because i could not get 1.8 to fly on rails like 1.7
I dont like to see iNav development getting worse

Pairan, 2018-01-02:
yes ... didnt have it with v1.7.3 ... allthough I do like the new OSD stuff in v1.8.x I remember v1.7.3 to be smoother ... Yet ... I will try to configure something when the weather is flyable! will keep you posted @Schalonsus

shellixyz, 2018-01-16:
Default settings may have changed between 1.7 and 1.8 you should check. About the buzzer issue are you sure your battery voltage doesn't go bellow the battery thresholds set in the `Configuration` tab ?

Pairan, 2018-01-16:
Hi shelli,

as I'm flying a 3s5200 LiHv it is pretty unlikely that it is a low voltage alarm. It's more like something that triggers the Buzzer constantly which then again isnt misconfigured (I double checked back then)

currently the s800 also starts to oscilate like in the video above. This is new! So I will have to run some checks on the cg and retry ... [you can see it here](https://youtu.be/diywJ1EdsZE?t=1m36s)

shellixyz, 2018-01-16:
If you are using LiHv it could be the issue because if iNav thinks your battery has one more cell than it really has then the battery alarm would be triggered all the time. Check that your max voltage is configured appropriately.

Pairan, 2018-01-16:
nah, did set it up to understand the cell voltage ist 4,5 max! ... no issue here as I else would not be able to arm it due to error ... if it refers to a change in the Cog then I will find it and report back in

shellixyz, 2018-01-16:
With the default settings all my wings are oscillating too but I tweaking the `ALTHOLD` relevant setting solved the issue.

shellixyz, 2018-01-16:
I lowered `nav_fw_pos_z_p` and raised `nav_fw_pos_z_d`. I also lowered `inav_gps_delay`

Pairan, 2018-01-16:
huh ... I will try that :) thanks for the hint

digitalentity, 2018-01-21:
@Pairan do you have any results yet?

Pairan, 2018-01-22:
@digitalentity ... still on it as flying wasnt possible at time, when I was ready! I'll still on it ... ( ... and I hate to say that!)

btw: ... nice to read you (again) ... it felt a little like everybody was gone here, and I'm really looking forward to try the optical flow code of yours

Pairan, 2018-02-13:
Just managed to give the settings a try:

What I configured
```
# set nav_fw_pos_z_p
nav_fw_pos_z_p = 15
Allowed range: 0 - 255

# set nav_fw_pos_z_p = 10
nav_fw_pos_z_p set to 10

# set nav_fw_pos_z_d
nav_fw_pos_z_d = 0
Allowed range: 0 - 255

# set nav_fw_pos_z_d = 5
nav_fw_pos_z_d set to 5

save
```

But that made the behaviour even crazier! I raised the `nav_fw_pos_z_p=17` for the next run! Will report back in, video is present.

shellixyz, 2018-02-13:
For reference I'm using these settings on a Micro Skyhunter:

```
set nav_fw_pos_z_p = 30
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 20
```

Pairan, 2018-02-13:
I will try these next time I get the chance - assume it to be this week somewhere :)

Pairan, 2018-02-14:
we are getting there - just took the chance and tried with these settings of yours and it worked better ... less waving, more chance to get home on RTH or circle around! It fights a bit against the wind but it managed to get home ... so yes, this was a good direction to get that issue sorted!

can anybody explain what exactly these settings do? I would like to be able to explain it if I am asked :)

shellixyz, 2018-02-14:
Good news :)

The altitude controller is basically just like any other PID controller. The P setting controls the pitch Proportionally to the altitude error. It makes the craft react to the present altitude error. The I setting controls the pitch proportionally to the accumulated (Integrated) altitude error. It makes the craft react to past altitude errors it complements the proportional control to cancel the offset inherent to P controllers when the error is small. The D setting controls the pitch proportionally to the speed of evolution of the altitude error (Derivative). It tries to "predict the future" to prevent overshoots/undershoots and improve the reaction time of the controller.

Good videos to understand PID controllers:
- https://www.youtube.com/watch?v=0vqWyramGy8
- https://www.youtube.com/watch?v=VVOi2dbtxC0

shellixyz, 2018-02-14:
Also realize than tuning these parameters is the last part to ALTHOLD tuning. For ALTHOLD to work best your board pitch alignment must first be tuned so that with throttle = `nav_fw_cruise_thr` and  without pitch input your wing does not lose nor gain altitude.

Pairan, 2018-02-18:
Okay ... while im still tweaking, i guess we can close this! It's something that simply has to be adjusted on a oscillating wing. Nothing more, i guess!

Got a video ready for the others who may still search for a clue on HOW to get it sorted:

https://youtu.be/LbM2FQ2mHlg

Schalonsus, 2018-02-18:
I tried this half a year ago to tune it with alt pid but did not solve my issue completely. It got better but not perfect. Tried every possible value combination, this was also topic at the last FPV-Treff meeting last year, many people had the same problem. It is definately a fact that this issue was not that big in previous iNav versions. I myself dont use 1.8 because control got rubbish, flashed 1.7 back to my race wing for better and more precise control. It is sad to see that the code is getting worse in some niche.

shellixyz, 2018-02-18:
@Pairan Where did you see this in the wiki ? It is not up to date the d term is used by the ALTHOLD controller.

Pairan, 2018-02-18:
@shellixyz :
here => https://github.com/iNavFlight/inav/wiki/Developer-info#pid-regulators-in-posholdrthwp-modes-xy-controller

Good to know ...

Pairan, 2018-02-18:
I guess i succeded get the S800 Sky Shadow to do PosHold safely. As it was pretty bumpy up there, I'll recheck on a calmer day. But the video leaves the impression, that it had to fight the wind, not oscillations! This is a good sign for me. Altitude varied +- 0.2m in the OSD ...

shellixyz, 2018-02-19:
It is the first time I'm seeing this page. I will ask if it needs to be updated or if I'm wrong but I red the code I don't think I am.

Very nice 0.2m is almost nothing !

Pairan, 2018-02-20:
yea ... hope to recofirm that as next flight will have to be on a calmer day ... AND WARMER DAY :D But knowing it will return safe and stay in the air is a good sign

closed 2018-05-27 by stale[bot]