iNavFlight/inav #2509: adc reading of rssi not stable 1.8.0

jnb81, 2017-11-26:
Hi, I've just flashed new FW 1.8.0 into my omnibus f4 pro and got rssi value changing from 100-39%
pretty default settings except of set rssi_scale = 39 (the same setting was working pretty well in previous release 1.7.3, and I was happy with it)
I'm using low pass filter from RX to convert RSSI PWM into analogue value (0-3.3V) the same set up was working fine with previous (stable) release.
Radio (TX) was all the time 0.5m from RX
video link:
https://youtu.be/O8SBl4JbruU
[damp-rssi-scale-39-1.8.0.txt](https://github.com/iNavFlight/inav/files/1504345/damp-rssi-scale-39-1.8.0.txt)

JingGeocacher, 2017-11-27:
my BeeRotor F4 with iNav 1.8.0 have same problem
https://www.rcgroups.com/forums/showpost.php?p=38663253&postcount=16720

digitalentity, 2017-11-27:
@jnb81 can't reproduce the issue. RSSI is received correctly, no random jumps.

jnb81, 2017-11-27:
@JingGeocacher
I had no jumps when default set rssi_scale = 100 
the value is constant 100 regardless of radio tx position (guess it was jumping from 100-100)
When I change the set rssi_scale to any number it will jump from 100 to that particular number and nothing else. 

@digitalentity 
Interesting....  did you also change default value of set rssi_scale (to something else than 100)?
I did the test twice and can not figure out what is wrong the only change is FW. The same configuration +hw works fine in 1.7.3

digitalentity, 2017-11-27:
@jnb81 no, I did test with rssi_scale=100. Ok, this narrows it down, will investigate.

digitalentity, 2017-11-27:
Fixed in #2510

closed 2017-11-27 by digitalentity

umdpru, 2017-12-06:
I have just wired up a flight controller running iNav 1.8 and I also get the jumping erratic RSSI reading if I change it to anything other than default 100.

digitalentity, 2017-12-06:
It's fixed already. Will be included in 1.9