iNavFlight/inav #2511: BetaFlight F3 Controller and Inav Firmware

tonysmartantzoulatos, 2017-11-27:
I have a brand new BetaFlight F3 controller from Banggood and i tried flashing it with Ivan Omnibus but also other firmware version.
Ivan locates my controller but does not flash and says it cannot start bootloader. It i connect my controller on DFU mode, it flashes omnibus (i tested this version because i read from a couple of people that it worked for them) but then my PC (win7) does not recognise the devide.
Connect again through DFU and reflash BetaFlight and it all works.

Is there a way to flash Ivan on my Betaflight F3? This is the board i have:
https://www.banggood.com/BETAFLIGHT-30_5x30_5mm-F3-Flight-Controller-Built-in-OSD-PDB-SD-Card-BEC-and-Current-Sensor-p-1162891.html?rmmds=buy&cur_warehouse=CN

Thank you in advance

shellixyz, 2017-11-27:
Omnibus should be the right target for your flight controller. Remember that should not flash an incompatible version because there is a chance you can damage the CPU or/and other devices on the board.

iNav is probably working fine. You should check the device manager to see if you can see the device and if drivers are properly installed.

Please remember the issue tracker on github is not for user support. Consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

tonysmartantzoulatos, 2017-11-27:
Hello shellixyz and thank you for your reply.
now that the board is with betaflight it can find it as com24. if i open inav configurator and try to flash with Omnibus from 1.3 up to 1.8 it simply does not flash (this is what i actually wanted to say on my original ticket). If i disconnect the board and press the button to connect as DFU , then it flashes the board and stops recognizing it. Any ideas on this? This is the only reason why i posted here since i already talked with people on the inav FB page and it all came to placing an issue here.

shellixyz, 2017-11-27:
After flashing even if you don't see the COM port in the configurator you should be able to see it listed in the Windows device manager. [https://www.drivereasy.com/knowledge/a-common-and-easy-way-to-access-device-manager-in-windows/](How to open Windows 7 device manager).

Are the LED indicators on your board flashing rapidly for some time when powering on after programming iNav ?

tonysmartantzoulatos, 2017-11-27:
As soon as board flashes (in DFU mode only) the win7 stops finding it. It says unknown devide. I tried updating driver, also tried zadig. Only when i put the board on DFU and take to betaflight and run beta once again. 
When flasing Omnibus on inav after the flash, the board seems to flash 2 slow and 3 quick (from the times i counted, it seems like 3).

tonysmartantzoulatos, 2017-11-27:
[image]

digitalentity, 2017-11-27:
@tonysmartantzoulatos looks like it's incompatible with OMNIBUS target. Which target do you flash on it to get it working in Betaflight?

tonysmartantzoulatos, 2017-11-27:
[image]

digitalentity, 2017-11-27:
Yes, that's different target which we don't have support in INAV yet. See also https://github.com/iNavFlight/inav/issues/1389

digitalentity, 2017-11-27:
Closing. Duplicate of https://github.com/iNavFlight/inav/issues/1389

closed 2017-11-27 by digitalentity

shellixyz, 2017-11-27:
@tonysmartantzoulatos I tried to add support for this board, feedback is welcome in PR #2514. You can find the code here https://github.com/shellixyz/inav/tree/BETAFLIGHTF3_target_complete

tonysmartantzoulatos, 2017-12-04:
an image would be appreciated for me to test on. Thank you

tasosman, 2018-06-08:
hello guys because i destroy the one pad for for one servo how can i remap this to strip led or for second motor to i cant use??i destroy the m3 servo input and now i cant fly