iNavFlight/inav #2521: Quadcopter shoots up at full throttle (Motors at 100%) with throttle stick centered

0120shrey, 2017-11-28:
SP Racing F3
INAV/SPRACINGF3 1.8.0 Nov  1 2017 / 06:21:53 (912d1315)

After adding a new GPS module, I was testing my quadcopter in POSHOLD mode. After holding it's position roughly (1 meter radius) for about 2-3 minutes, it shot straight up, even though my throttle stick was centered. Also, ALTHOLD was enabled throughout the flight, with maximum ascent rate set to 100cm/s, which should've prevented the quadcopter from ascending so fast. The blackbox log shows rcData[3] (which is throttle) at about 1500, which can then be seen reducing as I reduced the throttle. However, the parameter rcCommand[throttle] stays at 100%. As a last ditch effort, I dropped the throttle stick to the lowest position and disarmed the quadcopter, after which the motors stopped and it fell to the ground. 
After repairing the frame, the quadcopter flew perfectly in ALTHOLD (No GPS).

* _Steps needed to reproduce the problem_
I have no idea on how to reproduce it.

* _Expected Results_
Quadcopter should not have shot up. And after it did, it should've descended as the stick was below mid throttle and ALTHOLD was enabled.

* _Actual Results_
Motors stayed at 100% until the quadcopter was disarmed.

Dump
https://pastebin.com/vazhdgg9

[blackbox_log_2017-11-28_151646.zip](https://github.com/iNavFlight/inav/files/1510076/blackbox_log_2017-11-28_151646.zip)

digitalentity, 2017-11-28:
During that flight your GPS was returning jumpy and inconsistent data. I.e. jumping between (16.7772160, 0.0002072), (25.1582223, 0.0008216) and (28.6420476, 77.3230037) coordinates. Altitude was jumping from 2m to 1km in a fraction of a second.

Through most of the flight position estimator was able to filter out these bad readings, but at some point one got through and totally messed up the altitude calculation. INAV was thinking that copter is falling at extremely high speed and raised throttle trying to compensate.

Since the code always uses climb rate reported by GPS, it is succeptible to bad GPS data. I don't think we can do much about this - if GPS module reports data as valid and accurate we consider them usable.

0120shrey, 2017-11-28:
My ALTHOLD works very well without the GPS, using just the barometer. If I set the GPS weight to 0 for height calculation, will that avoid such a situation again? And also, are such readings from the GPS normal, or should I consider it a faulty module? It's a UBLOX M8N

digitalentity, 2017-11-28:
You can disable using GPS climb rate for altitude hold by setting `inav_w_z_gps_v` to zero.
Such data from GPS module is not normal. I'd say you either have some GPS jamming device operating nearby or a faulty module.

0120shrey, 2017-11-28:
Thank you! Sorry for putting it here, I thought it's a software issue.

closed 2017-11-28 by 0120shrey

digitalentity, 2017-11-28:
@0120shrey no worries, that's why we have "Support" label here.