iNavFlight/inav #2525: INAV OSD font upload on OMNIBUS F3 AIO hangs

krugs44364, 2017-11-28:
I am unable to upload OSD fonts on my Omnibus F3 AIO.
Rolled back firmware to 1.6.0 and it works.
Stepped forward firmware up to 1.7.2 and it works.
On 1.7.3 and 1.8 it just hangs.
So a work around is to flash 1.7.2,, load the large or bold font and then re-flash 1.7.3 and the new font stays there.
A sencondary issue with OSD on 1.8 is that the battery voltage indicator does not display correctly. For say 12.6V it displays 1, a space and a graphic character that I cant make out. 
So apart from fixing the ghosting issue which is great, OSD on 1.8 is useless since battery voltage is one of the main items I am interested in.
Is there a solution or do I just stick with 1.7.2 or 1.7.3?

giacomo892, 2017-11-28:
Are you using macOS? If so, it is a known bug in serial adapter driver

krugs44364, 2017-11-29:
No, I'm using windows and it behaves the same on 7 and XP. Haven't tried it on 10.
Just to be clear, the osd is the integrated osd on the Omnibus.

CrashAndBash, 2017-12-05:
I have exactly the same issue with the integrated OSD in INAV 1.8 on an OmnibusF4Pro. Using Windows 10. Voltage and current are unreadable. 1 a space and some weird blurred graphic. Tried changing the camera from NTSC to PAL and vice versa, fiddling with fonts but they remain unreadable. As I already have battery telemetry on my radio I think I will stick with 1.8 and wait for a fix and not rollback to 1.7.3

CrashAndBash, 2017-12-07:
FYI. I resolved unreadable OSD by uploading bold fonts. Afterwards everything became readable. Only issue now is artificial horizon banks the wrong way. Correct on Setup tab, wrong on OSD.

krugs44364, 2017-12-08:
Thanks. I will try the bold font.

digitalentity, 2017-12-08:
Can't reproduce. All fonts upload fine with 1.8 on my OMNIBUS F3

DzikuVx, 2017-12-27:
Issue seems to be stuck, closing. Thanks

closed 2017-12-27 by DzikuVx