iNavFlight/inav #2528: SP Racing Magnetometer Readings Stuck on 1 of 3 Axis (iNav)

Kyle5G8, 2017-11-29:
Hello all

I've just got the SP racing f3, but am having troubles with the magnetometer.

note: i2c speed set to 200KHZ to avoid i2c errors.

relevant things I've noticed

   In the setup tab where all the calibrations are done, rotating the board rotates the virtual quad copter correctly... but even when holding it still, it drifts back to where it was. It always goes back to where it started with great precision. 

   Now on to the sensors tab. When looking at the graph for the magnetometer the first time, moving the board caused changes on the x and z lines while the Y flat lined. After re flashing version 1.8 just in case, the same issue  occurred, but this time the y and z lines moved while the x line flat lined. I believe this shows it is not a bad sensor, as all axis are capable of functioning.  

   Lastly, when calibrating the mag and then going into CLI to view the magzero_xyz variables, the axis that flat line as mentioned before was a number in the tens of thousands while the other two that moved were more normal compared to what I've seen online.

This is a strange issue, and I hope it can be fixed. Any tips?

teckel12, 2017-11-29:
@Kyle5G8 I've noticed with clone SP Racing F3 FCs the orientation of the magnetometer is different (probably due to a clone magnetometer).  You probably need to flip/rotate the magnetometer settings till you find the correct side and orientation (there's only 8 combinations, so just try them all till you find one that works correctly).

Kyle5G8, 2017-11-29:
Ok I just went through all the possible magnetometer alignments.

Results: One of the axis still stayed locked wile the other two moved, with a different axis being stuck each time.

Kyle5G8, 2017-11-30:
After having the same problem across betaflight, CF, and inav, i do think its a hardware issue now. I'm getting an external i2c mag, and mounting on a mast. Closing

closed 2017-11-30 by Kyle5G8