iNavFlight/inav #2531: Accelerometer not working correctly in 1.8.1 at #2527 iNavFlight/de_fix_acc_calculation

giacomo892, 2017-11-29:
I flashed my OMNIBUS F3 with latest development  77a2e859e Merge pull request #2527 from iNavFlight/de_fix_acc_calculation and I found out that accelerometer isn't working right when the board is upside down.

The acc is calibrated and I tried even to recalibrate it on 1.8.1
No special configuration applied. Just stock + calibration.

The accZ should indicate -1G when upside down but as soon I rotate the board all the axis tend/go to zero.

Working 1.8 build

[image]

Not working 1.8.1 build:

[image]

digitalentity, 2017-11-29:
Appears to be a harmless type conversion error in MSP layer. Fixed in https://github.com/iNavFlight/inav/pull/2532

closed 2017-11-29 by digitalentity

giacomo892, 2017-11-29:
I'll try it later. Thanks for the fix