iNavFlight/inav #2533: Request: flip32 second softserial.

sashan13, 2017-11-29:
### Board and Version
flip32 lite - the one with mpu6050 only. 
### Behavior
This a new feature request!

As the title goes. I'm out of UARTs, and considering the fact that F4 processor has lots of computing power in spare I guess it can handle second softserial. 

I've got it as follows: 
uart1 - s.bus
uart3 - pins occupied by i2c.
uart6 - GPS
softserial1 - smart port telemetry

And now I'm out of UARTs for OSD. 

Reasonable? ;)

DzikuVx, 2017-11-30:
Yes, but where would you connect TX and RX of sencond SoftwareSerial? There are no reusable pins for us on this board

sashan13, 2017-12-01:
There is "ch2" pad free - under "UART3 TX" arrow on the picture. 
There are also PB12 and PB13 pads on the bottom - routed to respective pins 33 and 34 on the processor.  IMO using them would be the best. Yes, I've just checked it with multimeter, they are indeed routed that way.

closed 2018-05-27 by stale[bot]