iNavFlight/inav #2539: region `RAM' overflowed by 264 bytes

raul-ortega, 2017-12-01:
I apologize if this is not the correct place to post this. I'm trying to build OMNIBUS target but I get an error message when linking. However if I build OMNIBUSF4 the ends with success. Thanks so much.

%% settings.c
Linking OMNIBUS
c:/devtools/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: obj/main/inav_OMNIBUS.elf section `.bss' will not fit in region `RAM'
c:/devtools/bin/../lib/gcc/arm-none-eabi/4.9.3/../../../../arm-none-eabi/bin/ld.exe: region `RAM' overflowed by 264 bytes
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:981: obj/main/inav_OMNIBUS.elf] Error 1

giacomo892, 2017-12-01:
I think your compiler is too old, move to a newer version.

digitalentity, 2017-12-01:
@raul-ortega we use GCC version 6.3 or higher:
```
$ arm-none-eabi-gcc --version
arm-none-eabi-gcc.exe (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]
Copyright (C) 2016 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

OMNIBUS target builds perfectly
```
Linking OMNIBUS
arm-none-eabi-size ./obj/main/inav_OMNIBUS.elf
   text    data     bss     dec     hex filename
 213190    9860   38308  261358   3fcee ./obj/main/inav_OMNIBUS.elf
arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/inav_OMNIBUS.elf obj/inav_1.8.1_OMNIBUS.hex
Building OMNIBUS succeeded.
```

raul-ortega, 2017-12-02:
It is true, too old. I'm going to try and tell you...

raul-ortega, 2017-12-02:
I've installed the latest toolchan version and now everything is ok.
Thanks so much.

closed 2017-12-02 by raul-ortega