iNavFlight/inav #2540: FC / Raspberry / Mavlink / LTM

diyquad, 2017-12-01:
Hi,

I'm having troubles to send data from my FC to a raspberry.
I only want to get the telemetry data in the raspberry, so i can play with it in python.
(like gps position, altitude, flight mode etc.) 
Here is what i have:
- RPI 2 (and 3)
- FC Omnibus F4 v3
I tryed 2 things:
1) plug the FC & RPI in USB
2) Use the GPIO & UART1 of FC (RX/TX)

But no matter it send no data to the RPI.
I check MSP and/or Mavlink (also tryed LTM) on the Inav Port Configuration, but doesn't change anything: no data.
When i launch Inav directly from the RPI (with Chromium), it works, so i know the USB connection si OK, and i should be able to get all infos, but no

Anyone have an idea how to do it?
Thanks for help

diyquad, 2017-12-01:
Hey 
Thanks for quick answer.
Ok good to know, only when armed.
Is their a way to get the data anytime?
(only with msp? or telemetry?)

thanks again

DzikuVx, 2017-12-01:
Yes, use an UART that has only "Telemetry" function assigned to it

diyquad, 2017-12-01:
Great will try this today and let you know if it work.

diyquad, 2017-12-01:
Still one question:
Their is not "telemetry" option.
What should i use? FrSky? HoTT? LTM? Mavlink? 

Thanks again

DzikuVx, 2017-12-01:
Whichever you want and RPi can underdtand

closed 2017-12-07 by DzikuVx