iNavFlight/inav #2541: Mixer: FLYING_WING servo deflection problem

Martijn666, 2017-12-01:
output number 4 (right elevon) has more deflection (about 200%) on ROLL and PITCH through input from transmitter. When in any stabilized mode, the deflection on both surfaces is the same. 

Could this be a problem in FLYING WING mixer?

`
# version
# INAV/OMNIBUS 1.8.0 Nov  1 2017 / 06:14:28 (912d1315)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 -100 0 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
[... 478 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set tpa_rate = 33
set tpa_breakpoint = 1300`

DzikuVx, 2017-12-01:
This is line, nothing to worry about. One axis has only 50% authority for one servo

Od: Martijn666
Wysłano: piątek, 1 grudnia 2017 21:54
Do: iNavFlight/inav
DW: Subscribed
Temat: [iNavFlight/inav] Mixer: FLYING_WING servo deflection problem (#2541)

output number 4 (right elevon) has more deflection (about 200%) on ROLL and PITCH through input from transmitter. When in any stabilized mode, the deflection on both surfaces is the same.
Could this be a problem in FLYING WING mixer?
`
version
INAV/OMNIBUS 1.8.0 Nov 1 2017 / 06:14:28 (912d131)
resources
mixer
mixer FLYING_WING
mmix reset
servo
servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 -100 0
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
servo 7 1000 2000 1500 100 -1
servo mix
smix reset
feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD
beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 465 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 100
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 35
set pitch_rate = 9
set yaw_rate = 20
set tpa_rate = 33
set tpa_breakpoint = 1300`

MiroRadenovic, 2017-12-11:
I don't know if this is also related, but i'm having a problem with a really **limited servo deflection**.
even in pass through mode, the servo movement is really limited.
I can confirm that all my transimtters outputs are all calibrated (from 1000 to 2000)

here is a video of the problem: https://www.youtube.com/watch?v=MUQ8RhwwWiw

# version
# INAV/SPRACINGF3 1.8.0 Nov  1 2017 / 06:21:53 (912d1315)

# resources

# mixer
mixer FLYING_WING

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 -100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
[... 431 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

shellixyz, 2017-12-12:
@MiroRadenovic In the motors tab you can see what are the minimum/maximum values reached for each servo. With the default flying wing mixer full pitch should give you half travel as should full roll. You should get full deflection up on the right elevon when pitching up and rolling right for example.

Martijn666, 2017-12-12:
Even though the servo travels have been set identically, I get about 12mm deflection on the right surface and 9mm on the left surface. Both in Pass through aND assisted. I am an experienced fixed wing pilot.

MiroRadenovic, 2017-12-12:
Dear @shellixyz ,
thank you very much for your feedback. with your inputs i have tested everything works correctly. 
I can get full deflection (2000) with full pitch and full roll.
@Martijn666  sorry for my wrong input in your issue.

shellixyz, 2017-12-12:
@Martijn666 Did you check the servo outputs in the motors tab ? Looks like a hardware issue to me. What makes you think it is software related ? You should check with a servo tester the max deflections of your servos. My guess is some left/right difference in the linkages or positions of the servo horns or a cosine error because you compensated linkages length difference by rotating the horn on the servo or even a defective servo is causing that. Also be sure to select the right protocol for your servos. All servos are not able to cope with faster than 50Hz update rate.

DzikuVx, 2017-12-18:
Please bear in mind, that currectly configured flying wing should not reach full aileron deflection on full pitch or full roll. If that would be the case, servo would be saturated with only one axis and there were no room of other. I'm closing this one for now

closed 2017-12-18 by DzikuVx