iNavFlight/inav #2543: Matek F405-AIO board I2C do not work , firmware was build from the latest develop branch source

sincoder, 2017-12-02:
then I flash 1.8 and 1.7.3 ，I2C works will

digitalentity, 2017-12-02:
What devices do you have connected to I2C bus?

digitalentity, 2017-12-03:
This is strange. I would understand if `MATEKF405OSD` would fail because it uses enumalted I2C, but I don't see any issue with the code related to `MATEKF405` target. Can somebody in posession of a logic probe and MATEKF405-AIO board verify the activity on I2C bus?

@sincoder can you also provide a complete dump of non-working setup?

sincoder, 2017-12-03:
mbp:inav xx$ git branch
* development
  master
mbp:inav xx$ make TARGET=MATEKF405
settings.yaml -> settings_generated.h, settings_generated.c 
%% serial_softserial.c 
%% serial_uart_stm32f4xx.c 
%% blackbox_io.c 
%% cms.c 
%% cms_menu_blackbox.c 
%% cms_menu_builtin.c 
%% cms_menu_imu.c 
%% cms_menu_ledstrip.c 
%% cms_menu_misc.c 
%% cms_menu_navigation.c 
%% cms_menu_osd.c 
%% cms_menu_vtx_smartaudio.c 
%% cms_menu_vtx_tramp.c 
%% gps.c 
%% gps_ublox.c 
%% gps_nmea.c 
%% gps_naza.c 
%% gps_i2cnav.c 
%% ledstrip.c 
%% osd.c 
%% crsf.c 
%% frsky.c 
%% hott.c 
%% ibus_shared.c 
%% ibus.c 
%% ltm.c 
%% mavlink.c 
%% smartport.c 
%% telemetry.c 
%% vtx_smartaudio.c 
./src/main/io/vtx_smartaudio.c:72:5: warning: missing initializer for field 'channelCount' of 'vtxDeviceCapability_t' [-Wmissing-field-initializers]
     .capability.channelCount = 8,
     ^
In file included from ./src/main/io/vtx_smartaudio.c:44:0:
./src/main/drivers/vtx_common.h:35:13: note: 'channelCount' declared here
     uint8_t channelCount;
             ^
./src/main/io/vtx_smartaudio.c:73:5: warning: missing initializer for field 'powerCount' of 'vtxDeviceCapability_t' [-Wmissing-field-initializers]
     .capability.powerCount = 4,
     ^
In file included from ./src/main/io/vtx_smartaudio.c:44:0:
./src/main/drivers/vtx_common.h:36:13: note: 'powerCount' declared here
     uint8_t powerCount;
             ^
%% vtx_tramp.c 
./src/main/io/vtx_tramp.c:57:5: warning: missing initializer for field 'channelCount' of 'vtxDeviceCapability_t' [-Wmissing-field-initializers]
     .capability.channelCount = 8,
     ^
In file included from ./src/main/io/vtx_tramp.c:34:0:
./src/main/drivers/vtx_common.h:35:13: note: 'channelCount' declared here
     uint8_t channelCount;
             ^
./src/main/io/vtx_tramp.c:58:5: warning: missing initializer for field 'powerCount' of 'vtxDeviceCapability_t' [-Wmissing-field-initializers]
     .capability.powerCount = sizeof(trampPowerTable),
     ^
In file included from ./src/main/io/vtx_tramp.c:34:0:
./src/main/drivers/vtx_common.h:36:13: note: 'powerCount' declared here
     uint8_t powerCount;
             ^
%% main.c 
%% version.c 
%% printf.c 
%% serial.c 
%% serial_uart.c 
%% cli.c 
%% config.c 
%% fc_core.c 
%% fc_init.c 
%% fc_tasks.c 
%% fc_msp.c 
%% fc_msp_box.c 
%% settings.c 
%% serial.c 
%% serial_4way.c 
%% serial_4way_avrootloader.c 
%% serial_4way_stk500v2.c 
%% rcdevice.c 
%% rcdevice_cam.c 
%% msp_serial.c 
%% ibus.c 
%% jetiexbus.c 
%% rx.c 
%% crsf.c 
%% sbus.c 
%% spektrum.c 
%% sumd.c 
%% sumh.c 
%% xbus.c 
%% uav_interconnect_bus.c 
%% serial_usb_vcp.c 
Linking MATEKF405
arm-none-eabi-size ./obj/main/inav_MATEKF405.elf
   text	   data	    bss	    dec	    hex	filename
 217082	  13312	  46788	 277182	  43abe	./obj/main/inav_MATEKF405.elf
arm-none-eabi-objcopy -O ihex --set-start 0x8000000 obj/main/inav_MATEKF405.elf obj/inav_1.8.1_MATEKF405.hex
mbp:inav xx$ 

then I flash the hex file 
got this
[image]

reflash 1.8 or 1.7.3 works will

digitalentity, 2017-12-03:
@sincoder "provide a dump" = open Configurator, go to CLI tab, execute `dump` and post output here.

sincoder, 2017-12-03:
Entering CLI Mode, type 'exit' to return, or 'help'

# dump

# version
# INAV/MATEKF405 1.8.1 Dec  3 2017 / 20:07:33 (b0933cb4)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
beeper RX_SET
[... 478 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 20
set yaw_rate = 20
set tpa_rate = 0
set tpa_breakpoint = 1500

#

stronnag, 2017-12-03:
using code quotes helps.

What compiler are you using; I see no warnings building this.

sincoder, 2017-12-03:
gcc-arm-none-eabi-4_9-2015q2  ,i follow the wiki ,install this version

stronnag, 2017-12-03:
Obsolete compiler, please use 6.3

digitalentity, 2017-12-03:
Please upgrade to gcc-6.1 or higher

sincoder, 2017-12-03:
OK  ，i will test it now

digitalentity, 2017-12-03:
Although I2C should still work with binaries built with ancient compiler. Unfortunately I don't own the AIO board to debug this.

sincoder, 2017-12-03:
Upgrade to gcc-arm-none-eabi-6-2017-q2-update ,  do not have any warning on compile ,but 
  the I2C  still do not work, The icons always red .

digitalentity, 2017-12-04:
Found a bug. Will be fixed soon

sincoder, 2017-12-04:
after switch to branch de_busdev_i2c , rebuild and flash , I2C  works now !

closed 2017-12-04 by sincoder