iNavFlight/inav #2545: lux4osd target

wx4cb, 2017-12-02:
under betaflight 3.2.2 there's a luxf4osd target, is that the same as the lux race? this is related to my other omnibus f4 I have that is using that target. I was going to try that as opposed to the previous issue i'm having in another thread, but this one's using a different target than that one is.

shellixyz, 2017-12-04:
At first glance it looks like LUXF4OSD is like OMNIBUSF4 without SD card but you probably can't use the iNav OMNIBUSF4 target without at least disabling the SD card support.

wx4cb, 2017-12-04:
@shellixyz thanks for the reply, i'll give that a look, but there was an issue on the betaflight side of it where because it has a dataflash chip and no SD card, the baro wouldn't work because it's not on i2c but SPI.

here's the thread on betaflights github where we managed to get both working. That being said, i haven't completely tried it yet but it looks like in the process the osd was broken.

https://github.com/betaflight/betaflight/issues/1582#issuecomment-348732067

closed 2018-05-27 by stale[bot]