iNavFlight/inav #2549: Feature request, MSP over Crossfire

putimir, 2017-12-04:
I'm not sure if this is already in development / or done, I think it would be beneficial to have MSP over Crossfire protocol functionality, since Crossfire is quite popular in the iNav community.

Thanks

fiam, 2017-12-04:
It's somewhat in progress. I have a local branch with MSP over CRSF working, but I want to clean it up by unifying some of iNAV and BF APIs before we merge it.

putimir, 2018-02-20:
Is this moved to 2.0? :(

Can I build it myself to test? Thanks

fiam, 2018-02-21:
@putimir Unfortunately, I haven't managed to find time to clean this up and submit it. I've just removed myself from the issue assignment, just in case someone else wants to step in.

shellixyz, 2019-03-02:
Added by #4094, available in 2.1.0, closing.

closed 2019-03-02 by shellixyz