iNavFlight/inav #2552: New Target Request - Matek F722

saliimam, 2017-12-05:
newest Matek f7 board - has a baro onboard, with i2c available plus 4 uarts.

I have one to test.  If I have time in the next few weekends I'd try to help, but figure get this request in.

digitalentity, 2017-12-05:
Myself and @Mateksys are already on it.

saliimam, 2017-12-05:
Thanks @digitalentity - should I close this? Or attach to the PR?  If you need testers, let me know - should be installing this weekend.

digitalentity, 2017-12-05:
No, don't close it please - we need to track the requiest anyhow. I should be able to provide a hex for you in a few days.

KnuckleUpFPV, 2017-12-06:
I am looking forward to this. I tried using betaflight but even with mateks firmware it has 3 i2c errors. I tried using betaflights firmware and i lose accel, baro and gps. I cannot get this board setup in my wing. I jumped the gun and purchased too soon. Im ready to test as soon as a hex is available. Please let me know what I can do. Im using openlrsng non inverted sbus on my f3 with inav. Just no baro. I just had some barometers delivered so going to add it. I will fly that 3d printed wing until the hex is ready. Thank you for building the hex.

Bobrcnut, 2017-12-06:
I'm also looking forward to this one.. ready to replace several of my old faithful F3 Deluxe FC's.

marwady, 2017-12-11:
I have just received my F722....kooks really nice and cant wait to try with iNav.

Patiently waiting for hex......

Antiheld86, 2017-12-20:
Also looking forward to this one! I already have the F722 build into my F550 flamewheel.Thank you fror you developement!

KnuckleUpFPV, 2017-12-22:
any update by chance?

jmillerfo, 2018-01-30:
I have a Matek F722 running on a new Quad Copter frame.  In BetaFlight version 3.2.4 all configured items appear to be running correctly including the magnetometer although I do get 3 I2C errors on boot.  This includes SmartPort Telemetry, Wireless BT on Serial 1, M8N GPS and Magnetometer, LED Strip,etc. When I load Matek's iNav 1.8.1 (1/24/2018) I note the following:
- Flashing via DFU works normally with only the USB connected but upon the reboot I have to connect the battery before it will boot.  I checked the 5v going to the receiver and it read 1.4v.  On BF it reads 4.8v.  It appears something is drawing a lot of correct such that the USB cannot provide sufficient amperage to power the F722.  The USB Port is capable of supplying 2.1A.
- When turning on the LED_STRIP either via the GUI or via CLI *feature LED_STRIP* and performing a reboot, I am not able to connect to the F722 either via USB or Serial Port 1.  The reboot appears normal but after a few seconds the activity LED stops flashing, no OSD output to FPV although the camera video is present (Camera is connected to F722).  On BF LED_STRIP is assigned to resource A15.  On iNav A15 is free but the LED_STRIP does not appear on the resource list.

digitalentity, 2018-01-30:
@jmillerfo see https://github.com/iNavFlight/inav/issues/2686

digitalentity, 2018-01-30:
Target merged to `development` closing.

closed 2018-01-30 by digitalentity