iNavFlight/inav #2553: ANYFC F7 BB log not working in 1.8

flyfisher2, 2017-12-05:
Just upgraded from 1.7 to 1.8 and have a problem with BB logging.

After the uprade using a new empty card, in the BB tab, instead of the normal message showing available space, I am getting "FILE SYSTEM STARTING".This creates a FREESPACE.E file of 0KB on the card and not surprisingly, no log folder or files can be created.

Initialising the card in 1.7 then using it in 1.8, new log files can be created in the LOGS folder but they are always shown as 4096KB long and basically contain garbage.

sambas, 2017-12-10:
F7 would need some porting from BF for SPI_LL driver to make SD card work more reliable.

flyfisher2, 2018-01-08:
Well, guess not.
Time to change to another type.

digitalentity, 2018-01-10:
@sambas when you would be able to look into this?

sambas, 2018-01-10:
I'm middle of moving my place, so might take a while.

digitalentity, 2018-01-10:
@sambas noted, thanks. Might have a look at it myself if I have the time. Is the issue fixed in Betaflight?

sambas, 2018-01-10:
Last time I check it was working ok.

digitalentity, 2018-01-10:
Ack. I'll have a look.

closed 2018-01-28 by flyfisher2

flyfisher2, 2018-01-28:
Retested BB log with 1.8 and it is now working correctly.
Only difference, this time I used a formatting tool from the SD Association and not the one included in WIN 10.
According to them, operating system formatting tools may not optimise SDHC cards.