iNavFlight/inav #2554: Difference between RC and Stabilization in SMIX

TNgineering, 2017-12-05:
Does anyone know the real difference between the RC roll/pitch/yaw and the stabilized roll/pitch/yaw on the SMIX section? My control servos seem to work fine with only using the stabilized set. What is the RC set used for?

DzikuVx, 2017-12-05:
Like name suggests:

* RC is what come from RC link. For example to move servos with sliders on RC radio
* Stabilized is what comes out of flight controller after stabilization (gyro and accelerometer) is applied

closed 2017-12-05 by DzikuVx

TNgineering, 2017-12-05:
Hi Paweł, thanks for your quick response. Could you please clarify a little further? Once I get this figured out I can update the wiki.

My tests show that using:
`smix reset`
`smix 0 4 5 100 0 #Rudder RC`
`smix 1 3 6 100 0 #Elevator RC`
`smix 2 5 4 100 0 #Aileron RC`
`smix 3 6 11 100 0 #Flaps Aux 11`

only allows input from the RC transmitter controls which is what I would expect. 

However, using:
`smix reset`
`smix 0 4 1 100 0 #Rudder Stab`
`smix 1 3 2 100 0 #Elevator Stab` 
`smix 2 5 0 100 0 #Aileron Stab`
`smix 3 6 11 100 0 #Flaps Aux 11`

allows the servos to move using both the stabilization AND the inputs from the RC transmitter. 

Are the stabilization outputs designed to include the inputs from the RC transmitter or are they meant to be separate? For example, if they were meant to be separate, I would then assume that I would have to combine the SMIX setup to be like this:
`smix reset`
`smix 0 4 1 100 0 #Rudder Stab`
`smix 1 4 5 100 0 #Rudder RC`
`smix 2 3 2 100 0 #Elevator Stab`  
`smix 3 3 6 100 0 #Elevator RC`
`smix 4 5 0 100 0 #Aileron Stab`  
`smix 5 5 4 100 0 #Aileron RC`
`smix 6 6 11 100 0 #Flaps Aux 11`

What's also weird is that I get noticeable servo clipping only in the YAW servo when using the combined smix setup but not in the PITCH or ROLL direction. Do you have any thoughts as why this is happening?

DzikuVx, 2017-12-05:
Stabilization uses data from both gyro and RC input. If you why any flight mode if INAV, you have to use 'stabilized'

TNgineering, 2017-12-05:
Ok so it sounds like it was intended to be that way. 

On another note, are only AUX channels 1-4 available for servo outputs using SMIX? These are the only ones listed on the wiki.

shellixyz, 2017-12-08:
Yes only AUX 1-4 are available for now