iNavFlight/inav #2556: omnibus f4 pro CH5 CH6 softserial move to M2 or M6

davidngrc, 2017-12-05:
two of my omnibus f4 pro CH5 CH6 is gone (got rip off)
And I found out many people have the same issue.

I am using this board in flying wing which only use M1, M3, M4.
I know M5 is use for LED for clones.
is it possible to add an CLI command to enable the softserial on M2 or M6?
or move the CH5 CH6 resource to M2 or M6?

my current setup is, sbus + gps + mag + sport on softserial
Thank you so much.

DzikuVx, 2017-12-05:
It is not possible to remap this via software configuration I'm affraid

davidngrc, 2017-12-06:
I see in betaflight, they can do sport on M5 https://github.com/betaflight/betaflight/wiki/Single-Wire-Software-Serial
and in iNav, the M5 can use for LED.
is that mean, I can modify the iNav source code, to config the softserial on M5 by just replacing the LED code as CH5 and CH6 code?

digitalentity, 2017-12-06:
You certanly can hack the code in any way you like. Resource remapping as in Betaflight is not available in INAV and likely won't be available any time soon.

davidngrc, 2017-12-06:
need some help, 
I following this guide to setup the compile environment. https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md
git the v1.8 source code, and compile it using command " make TARGET=OMNIBUSF4PRO_LEDSTRIPM5 "
the new firmware flash ok.
I then follow "FrSky SmartPort using SoftwareSerial" https://github.com/iNavFlight/inav/blob/master/docs/Board%20-%20Omnibus%20F4.md
the CH5 CH6 softserial work fine, LED strip on M5 also works.

I then delete the whole inav source code, and git again.
I change the softserial from CH5 and CH6 to M5 and M6, and the LED strip from M5 to M2 as shown below,

in \inav\src\main\target\OMNIBUSF4\target.h
`
//#define SOFTSERIAL_1_RX_PIN     PC8
//#define SOFTSERIAL_1_TX_PIN     PC9
#define SOFTSERIAL_1_RX_PIN     PA1
#define SOFTSERIAL_1_TX_PIN     PA8

//# define WS2811_PIN                      PA1
#   define WS2811_PIN                      PB1
`

in \inav\src\main\target\OMNIBUSF4\target.c
`
//    { TIM8,  IO_TAG(PC8),  TIM_Channel_3, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: small CH5 pad, not used as PWM, definition inherited from REVO target
//    { TIM8,  IO_TAG(PC9),  TIM_Channel_4, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: small CH6 pad, not used as PWM, definition inherited from REVO target
    { TIM8,  IO_TAG(PA1),  TIM_Channel_3, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: small CH5 pad, not used as PWM, definition inherited from REVO target
    { TIM8,  IO_TAG(PA8),  TIM_Channel_4, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: small CH6 pad, not used as PWM, definition inherited from REVO target

    { TIM3,  IO_TAG(PB0),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR                    | TIM_USE_FW_MOTOR }, // MOTOR_1
//    { TIM3,  IO_TAG(PB1),  TIM_Channel_4, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR                    | TIM_USE_FW_MOTOR }, // MOTOR_2
    { TIM5,  IO_TAG(PB1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_2

    { TIM9,  IO_TAG(PA3),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM9, TIM_USE_MC_MOTOR                    | TIM_USE_FW_SERVO }, // MOTOR_3
    { TIM2,  IO_TAG(PA2),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM2, TIM_USE_MC_MOTOR                    | TIM_USE_FW_SERVO }, // MOTOR_4
#if (defined(OMNIBUSF4PRO) || defined(OMNIBUSF4V3)) && !defined(OMNIBUSF4PRO_LEDSTRIPM5)
//    { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
    { TIM4,  IO_TAG(PB6),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM4, TIM_USE_LED                                            }, // LED strip for F4 V2 / F4-Pro-0X and later
#else
//    { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
#endif
//    { TIM1,  IO_TAG(PA8),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_6
};

`

but then I flash and test it, the softserial and LED does not work.
please help.

| Motor     | pin   |
| ----      | ----  |
| 1         | PB0   |
| 2         | PB1   |
| 3         | PA3   |
| 4         | PA2   |
| 5         | PA1   |
| 6         | PA8   |

davidngrc, 2017-12-07:
The fact is, I can not even make the LED change from M5 to M6 by changing the following
I know it is relate to TIM5 , TIM_Channel_2 and GPIO_AF_TIM5
but I have no idea what their meaning. (eg. what 2 channel mean, why the code below does not work...)
I know programming, but not family with this type of dev.
anyone help?

in \inav\src\main\target\OMNIBUSF4\target.h
//#   define WS2811_PIN                      PA1
#define WS2811_PIN                      PA8

in \inav\src\main\target\OMNIBUSF4\target.c

//  { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
    { TIM5,  IO_TAG(PA8),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
#if (defined(OMNIBUSF4PRO) || defined(OMNIBUSF4V3)) && !defined(OMNIBUSF4PRO_LEDSTRIPM5)
    { TIM4,  IO_TAG(PB6),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM4, TIM_USE_LED                                            }, // LED strip for F4 V2 / F4-Pro-0X and later
#endif
//  { TIM1,  IO_TAG(PA8),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_6
    { TIM1,  IO_TAG(PA1),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_6

shellixyz, 2017-12-07:
@davidngrc Timers are bound to particular pins. You can't just change pins you also need to adapt the timer and channel. You can find which timer and timer channel needs to be used with a pin in the MCU datasheet in the "Pinouts and pin description" section.

davidngrc, 2017-12-08:
@shellixyz thank you.
it seems I mess up my two other good board by just doing the above test.
in one board, the M1 and M2 do not get any signal.
in another board, the M3,M4,M5,M6 do not get any signal...
flash betaflight, then flash inav stock, no luck. @_@.

shellixyz, 2017-12-08:
No, using the wrong timer for a pin shouldn't damage the chip. To damage a LED either the LED's current limiting resistor had it's a value dropped (contamination on the board) or too high voltage was applied causing too much current to flow through the LED. Could just be a defective LED but it is unlikely. There is 8 light emitting diodes on your FC ?

davidngrc, 2017-12-08:
for the LED test, I am using a 5v 3A BEC, test report 4.98v, and using this external WS2812B LED  item.taobao.com/item.htm?id=523218029121
in one of the board, I was careful, that I only solder the signal pin for M1 to M6, and the sbus, PC usb power this board. but somehow, this board suffer the most damage, M1 to M6 don't work.

shellixyz, 2017-12-08:
I thought you were talking about the FC status LEDs. RGB LEDs don't have external current limiting resistors. The problem could come from ESCs with regenerative braking. If you don't have enough capacitance on the power rail it can generate big voltage spikes with enough energy to fry chips when braking.

davidngrc, 2017-12-08:
holy... I have 0 cap on my FC as I did not solder any 5V or GND or BAT pin, I only power it by PC USB.
the FC only connect to the RGB LED by signal wire, the LED 5v was provide by the external BEC.
the motor is also connect to the FC by the signal wire only, the power is coming from a 40A ESC which has one cap side it.
and the 5v BEC and 40A ESC are power by a 4s battery.

I guest what you are saying is, when the RGB LED smoke, the RGB LED somehow mess up, and the signal wire put a big current (or voltage? but the BEC is 5v, 5v to single pin will fry it?) to the FC, and fry that chip.
I just order 2 more FC... 

do you guys have some simple method of recording the signal coming from M1 to M6?
in iNav config, I see the data change when the x9d stick move, but the pin seems not outputting the correct value or not outputting anything.

shellixyz, 2017-12-08:
If the cap on the ESC is big enough and is low enough resistance you should be good. No need for adding a cap on the FC, the caps to prevent voltage spikes and noise from the ESC should be as close as possible from the ESC power input.

About the RGB LED yes if it is really messed up it could source or sink current via the input pin and fry an output stage or more of the MCU. I never seen it but it is totally possible. A STM32F3 MCU pin can only source/sink around 10mA MAX.

For recording the signal the simplest method would be to use an oscilloscope. But if you don't have one and still want to check roughly the output I guess you can just use a multimeter. Set the outputs tu use standard 50Hz PWM. If the output is disabled (no signal) you should read 0V. If it is at min (1000µs) you should read 0.165V and at max (2000µs) it should read 0.330V. You can convert the voltage to the output signal with this formula: output_microsecs = tension / 3.3 * 20000
If you read 0.213V => 0.213 / 3.3 * 20000 = 1291µs (29%)

davidngrc, 2017-12-09:
well, the working M pin is about 0.25v, and the not working M pin is 3.25v
thank you so much.

davidngrc, 2017-12-09:
but I don't get it, the following change should work, but it didn't.
I try to change CH5 pad to M2, CH6 pad to M6, ignore the RGB LED.
I kept the original pin channel and original TIM
I test use TIM_USE_ANY for M2 and M6, or keep the original TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO, both don't work.
I even try to put this in different position of the array, all don't work.
anyone know why?

in \inav\src\main\target\OMNIBUSF4\target.h
#define SOFTSERIAL_1_RX_PIN     PB1
#define SOFTSERIAL_1_TX_PIN     PA8

in \inav\src\main\target\OMNIBUSF4\target.c
    { TIM8,  IO_TAG(PC6),  TIM_Channel_1, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: UART6 TX, not used as PWM, definition inherited from REVO target
    { TIM8,  IO_TAG(PC7),  TIM_Channel_2, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: UART6 RX, not used as PWM, definition inherited from REVO target

    { TIM3,  IO_TAG(PB1),  TIM_Channel_4, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_ANY }, // MOTOR_2, should be like CH5
    { TIM1,  IO_TAG(PA8),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM1, TIM_USE_ANY }, // MOTOR_6, should be like CH6

    { TIM3,  IO_TAG(PB0),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR                    | TIM_USE_FW_MOTOR }, // MOTOR_1
    { TIM8,  IO_TAG(PC8),  TIM_Channel_3, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: small CH5 pad, not used as PWM, definition inherited from REVO target
    { TIM9,  IO_TAG(PA3),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM9, TIM_USE_MC_MOTOR                    | TIM_USE_FW_SERVO }, // MOTOR_3
    { TIM2,  IO_TAG(PA2),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM2, TIM_USE_MC_MOTOR                    | TIM_USE_FW_SERVO }, // MOTOR_4
#if (defined(OMNIBUSF4PRO) || defined(OMNIBUSF4V3)) && !defined(OMNIBUSF4PRO_LEDSTRIPM5)
    { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
    { TIM4,  IO_TAG(PB6),  TIM_Channel_1, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM4, TIM_USE_LED                                            }, // LED strip for F4 V2 / F4-Pro-0X and later
#else
    { TIM5,  IO_TAG(PA1),  TIM_Channel_2, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM5, TIM_USE_MC_MOTOR | TIM_USE_MC_SERVO | TIM_USE_FW_SERVO }, // MOTOR_5
#endif
    { TIM8,  IO_TAG(PC9),  TIM_Channel_4, 0, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_ANY }, // Connected: small CH6 pad, not used as PWM, definition inherited from REVO target
};

| Motor     | pin   |
| ----      | ----  |
| 1         | PB0   |
| 2         | PB1   |
| 3         | PA3   |
| 4         | PA2   |
| 5         | PA1   |
| 6         | PA8   |

shellixyz, 2017-12-09:
0.25V is good it means the output is working and outputting 1500µs pulses. 3.25V means the output is stuck high and is maybe dead.

I don't know much about softserial but I think you should use TIM_USE_ANY for the pins. Please give me the model of the MCU on your board. STM32F4XX

davidngrc, 2017-12-09:
it is a omnibus f4 pro clone board, using ARM STM32F405

shellixyz, 2017-12-09:
You could try this: [inav_1.8.0_OMNIBUSF4PRO.hex](https://drive.google.com/open?id=1-0RPfARXhcl_aME9tNAKPrTOJVYpVyFm)
I tried to move SS RX on M5 and SS TX on M6. No guarantee.
The code: https://github.com/shellixyz/inav/tree/OMNIBUSF4_move_softserial_to_M5M6

davidngrc, 2017-12-09:
it seems, the code does not work.
enable softserial in inav, using default flying wing setting.
try both telemetry_inversion = OFF and ON, don't work.
try swap M5 and M6 pin (I am using 1k resistor), don't work
the X8R only show rssi and rxbt 2 sensor in x9d+

seeing the compare code, it is clean
https://github.com/iNavFlight/inav/compare/master...shellixyz:OMNIBUSF4_move_softserial_to_M5M6

could anyone else double confirm the hex @shellixyz  provided is working or not?

shellixyz, 2017-12-09:
Maybe this will work better: [inav_1.8.0_OMNIBUSF4PRO.hex](https://drive.google.com/open?id=1UVFT2UVzO9umkUUd2VNUuNuYXqm_DcZA)

davidngrc, 2017-12-09:
same, no luck. 
telemetry_inversion is on
swap M5 and M6 pin, no luck.

jcaamnes, 2017-12-14:
i also have a omnibus f4 pro clone. i can test this tomorrow.

davidngrc, 2017-12-19:
@jcaamnes may I know what is your test result?

zector555, 2017-12-19:
Id like to get this working as well.  Thanks

zector555, 2018-05-13:
The hex that shellixyz posted i believe is a fix for now.  It would be nice to see this as maybe a option for this board?

shellixyz, 2018-05-13:
Please check if this build is working and report: 
[inav_1.9.1_OMNIBUSF4PRO_SSM5M6.hex.zip](https://github.com/iNavFlight/inav/files/1998779/inav_1.9.1_OMNIBUSF4PRO_SSM5M6.hex.zip)

If it is working I will make a PR to include this new target in next release.

shellixyz, 2018-05-13:
MOTOR5 pad = SOFTSERIAL1 RX
MOTOR6 pad = SOFTSERIAL1 TX

davidngrc, 2018-07-07:
Sorry for the late reply, I was not getting any notification for this.
I use a new board test, first test the original CH5, CH6 softserial working, then swtich to 
CH6 -> MOTOR5
CH5 -> MOTOR6 (1k resister to MOTOR5)
It works. thank you.

could you make this into iNav 2.0 or show us how you did it?

davidngrc, 2018-07-07:
@_@, just get my old board which rip CH5 and CH6 pad, and done the previous test in this thread.
it seems my old board MOTOR5 and MOTOR6 also broken...
I was trying to use this board on a flying wing, now it left MOTOR2 is un-used, any way we could get a softserial out of one pin?
or is there any hidden pin in the omnibus f4 pro that can use?

davidngrc, 2018-07-17:
@shellixyz could you release a 2.0.0RC4 build for this? so that I can test it? thank you.

davidngrc, 2018-07-17:
@shellixyz 
or could you point me to your git, so that I can do a diff on the prod 1.91, and try to integrate it into iNav 2.0 by myself?
Thank you.

closed 2018-09-29 by stale[bot]