iNavFlight/inav #2557: Omnibus F4 (v2 pro / v3) RX serial connection will crash 0.5 to 1 sec after do a combination key

MichaelNTJ, 2017-12-06:
Hi, I am not sure the issue publish already or not  ,but still wish who can solve this problem.

My case:
I have 2 board of Omnibus F4 v2 pro and Omnibus F4 v3 , I found out a big problem is that
  when I do a  combination key  from remote control  then (my board has a beep ) I can hear a 
  beep sound and the top of INAV the icon  RX serial connection (the icon like a paper clip) will from blue color  change  to gray color 0.5 to 1sec , if you are also in receiver setting page,  during  the time any 
signal control is not working too untill  paper clip icon back to blue color.

I don't know where the problem point so I did come testing
A..change INAV to betafight , but the problem is same.
B.testing  remote control and receiver but the problem is same.
so I guess  possible is FW problem ?
 
[Issue Vedio:]
the video link is not my but same problem with my issue ,https://youtu.be/zCr0NahWqH4

 [combination key]  :
1. a full throttle + a 95% up pitch  or (full pitch up)
2. a full throttle + a 95%up roll(full right)  or  -95%  up roll (full left) 
 
[remote control check]
FLYSKY FS-i6X + receiver FS-IA10 (10ch) ( type PPM)  > failed
FLYSKY FS-i6X + receiver FS-A8S (8ch)   ( type PPM)  > failed
DEVO10 + receiver RX-SUBS (12ch)  ( type SBUS)     > failed

[INAV Version]
INAV  Configurator: 1.8.1   (also I woukd like to now , where can install old version?

[FW]
OmnibusF4v3
Flight controller info, identifier: INAV, version: 1.7.2
Flight controller info, identifier: INAV, version: 1.7.3
Flight controller info, identifier: INAV, version: 1.8.0

OmnibusF4PRO
Flight controller info, identifier: INAV, version: 1.7.1
Flight controller info, identifier: INAV, version: 1.7.2
Flight controller info, identifier: INAV, version: 1.7.3
Flight controller info, identifier: INAV, version: 1.8.0

digitalentity, 2017-12-06:
Please consider reading the docs https://github.com/iNavFlight/inav/blob/master/docs/Controls.md

When disarmed:
Throttle up + pitch/roll to the sides  = board alignment trimming.

With your stick combination you are changing board configuration and board stops accepting commands while new configuration is saved. This is normal behavior.

closed 2017-12-06 by digitalentity

MichaelNTJ, 2017-12-06:
HI ,  Thank you for explain than.
But I am a little confuse is that  when my fly drone  on the sky some time I need to have a  full power with full roll or full  pitch , during this time the board also get a stick position function command like what you said. 
after drone fall down(few time already ) because of receiver not working.  so I dont't know how could we say that is normal behavior? 

May I ask how to  turn off stick position function before  drone fly on the sky?    or avoid  any stick position function  command happen  on the sky?