iNavFlight/inav #2558: add *new* magnetometer DA 5883(QMC5883) into the list of mag_hardware in future firmware.

chan2328, 2017-12-06:
as #2299 the solution is found and tested which is removing the magGain. is it possible to add *new* magnetometer DA 5883(QMC5883) into the list of mag_hardware in future firmware?

digitalentity, 2017-12-07:
QMC5883 is supported already. What you are asking for is support for poorly cloned HMC5883 compass. I'm not sure that removing magGain from HMC5883 driver is a proper way of doing this as it will break existing setups.

chan2328, 2017-12-08:
@digitalentity Sorry, May be my explanation is not clear, I mean that add a additional driver for particular magnetometer (DA 5883),  no need to change existing driver, because I have HMC5883 too, the existing driver for HMC5883 is good and works great.
after flight few days. DA 5883 did very well, so I suggest adding into the list of mag_hardware for future firmware. and we should understand that we need to take a risk when using this magnetometer (DA 5883). Trial and Error. 
Actually, I want to modify firmware by myself, but it is out of my knowledge, as there is no easy way to get into the coding, no tutorial i can follow step by step to create custom firmware, whatever  thank you very much.

shellixyz, 2017-12-08:
@chan2328 
Developer documentation exists:  https://github.com/iNavFlight/inav/tree/master/docs/development
Compiling a custom firmware is relatively easy. Once you installed the ARM compiler toolchain just download the iNav source code and execute `make TARGET=your_target`.

stronnag, 2017-12-08:
see also the fine wiki https://github.com/iNavFlight/inav/wiki/Building-custom-firmware

chan2328, 2017-12-09:
@shellixyz @stronnag followed the step, but it is not work. what should i do next? install 1, 2, 3 steps in RUBY windows?
[image]

chan2328, 2017-12-09:
I am using windows 7 64bit . should i install 64bit version of RUBY or else

parnz, 2017-12-09:
Use docker solution to build inav is super easy :)

shellixyz, 2017-12-09:
@chan2328 What Ruby installer did you use ? The installer downloaded from https://rubyinstaller.org/ (the URL from [Building in Windows](https://github.com/iNavFlight/inav/blob/master/docs/development/Building%20in%20Windows.md)) should be a GUI and automatically add Ruby to your PATH. The problem is that the installer you used did not add the path of your Ruby binaries to the PATH environment.

chan2328, 2017-12-09:
Ruby 2.4.2-2 (x86) [image]

i lost.........
[image]

shellixyz, 2017-12-09:
I'm not using Windows but I installed the same version along with Cygwin (all defaults) in a fresh VM and it is working.
[image]

chan2328, 2017-12-11:
@shellixyz I also follow the steps but there is no git completion, it is different.
[image]

shellixyz, 2017-12-11:
@chan2328 You don't need the bash completion for git just skip it.