iNavFlight/inav #2561: External magnetometer drift

Kyle5G8, 2017-12-07:
Hello all, I am having constant issues with the magnetometer in inav. I'm using the sp racing f3.

The problem, in short: in the setup tab, as soon as I stop rotating the quadcopter, it drifts back to its original heading. Heading readings have no correlation with actual direction pointed. 

I've tried all board alignments, calibrating, and have had this issue with both on board mag and se100 external gps and mag. Its driving me insane... I'm close to calling it a bad board. 

Any tips?

KiwiDavid, 2017-12-07:
Hi -  I think if you're powering only off the USB cable then the GPS won't get powered up and the GPS is providing power to the mag sensor.  The internal mag should power up ok.   Try powering with a flight battery (no props) and see if the mag sensor comes alive.
BTW the mag sensor is listed as a HMC5983 - I didn't see it specifically listed in the iNav mag sensor list but perhaps someone can advise if it's supported.

stronnag, 2017-12-07:
The HMC5983 works as a HMC5883. I use one and it works well in iNav.

Kyle5G8, 2017-12-07:
Yes it configures well. The on board mag was removed, and replaced with the external. My module does provide power to the gps and mag on USB power, all indicator lights on the module are on and I can see my gps fix. 

So what do you guys think? bad board? or is there some other setting i've missed? I've found out I did get stuck with a clone board, but of all the problems surrounding clones I didn't see this as one of them

KiwiDavid, 2017-12-07:
OK so the device is supported and you do have power to the sensor.   My SP Racing F3 doesn't power the GPS via the USB.
You mentioned you had the same issue with the on-board mag sensor.  Does the mag sensor symbol light up blue or remain red?   i.e. is it being recognized but not working or just not recognized?
Try forcing the I2C clock speed down to 100kHz?  Do you have access to a logic analyzer or scope?  Check to see if the SDA and SCL lines are idling high at around 3.3V.  They will pulse low with bus activity  so a multimeter may read an average value of 1.8-2V.   This at least shows the pullups are intact as nothing else on the bus can pull high.
Sorry I can't help much more.

TimMW1, 2017-12-07:
I had the same issue with my external compass with SP racing F3.  I had to change  the compass orientation in the config tab.  I believe I had to move mine to flip90.  I believe this happens because the compass is going one way but the gyroscope/accelerator is showing the opposite.  Also, the external compass is powered from battery not USB on my setup.  I am running at default 400kHz I2C.

Kyle5G8, 2017-12-07:
Interesting, I'll keep trying orientations although I have probably tried them all. 

More interesting findings:

The mag reads blue in inav, but turns red when held diagonally downwards at a very specific angle. I've already ruled out and ensured no faulty connections. I'm thinking maybe this flicker of red, which also happens when calibrating, throws the algorithm off (if mag goes briefly red during calibration) causing it to act strange. dump shows that magzero_(xyz)=(-1,0,0).

When spinning it around in setup tab, it acts normal and perfect through some parts of the unit circle but snappy and holding one heading through the other half. This throws the heading off permanently though. 

All is the same with or without battery, I have the device drawing power from the i2c plug that has a 5v rail powered up on usb.

Despite responding to mag readings in the setup tab, the graph in the sensors tab shows flatlines as if the sensor weren't there.

KiwiDavid, 2017-12-08:
Hi,
I'm not sure about powering the entire board and the GPS via the 5V USB line.  
This from the SPRacing F3 manual-

"I2C1 connector - Used for external sensors and OLED displays.
The SCL and SDA are 3.3v signals.
5.0v is always supplied via the on-board voltage regulators, even when powering via USB."

You may get away with it but the USB will be close to maximum I would have thought, especially with the GPS in acquisition mode.
Is it worth trying with a flight battery or have you already tried this?

Kyle5G8, 2017-12-09:
Thanks for everyone's help, it was a combination of things. Those who said it was alignment were correct, but another error was also not letting it calibrate. Apparently inav has a known issue with compasses sometimes, and three lines of code had to be edited before rebuilding a hex, flashing, and boom it works.

After half a day of staring at a screen of course...

closed 2017-12-09 by Kyle5G8

digitalentity, 2017-12-10:
@Kyle5G8 good to know that you figured it out. Yes, compasses are tricky - especially the HMC5883 ones. Original HMC5883/5983 is EOL and is no longer produced - almost all chips we are dealing with are clones which reproduce HMC API with random accuracy. I.e. the chip known as "DA 5883" may be detected either as QMC5883 (which it actualyl is) or HMC5883 (which it may be emulating). In the latter case some of the magGain-related API is not emulated correctly.

kbrennan1961, 2018-03-21:
Kyle any insight into these three lines of code and the solution?  I am having a similar problem. Thanks.

-Kevin

Apparently inav has a known issue with compasses sometimes, and three lines of code had to be edited before rebuilding a hex, flashing, and boom it works.

georgekucher, 2018-05-29:
Having the same issue. Described [here](https://github.com/iNavFlight/inav/issues/606#issuecomment-392614595)

georgekucher, 2018-05-29:
Was a bad calibration