iNavFlight/inav #2562: Clone Omnibus f4 pwm channels linked

BrianKoch1, 2017-12-07:
Three different cloned omnibus f4 pro boards all with the same issue.

General airplane config

I've tried all versions of omnibus f4 pro firmware.

All radio channels show correct movements in receiver tab.
All servo outputs show correct in motor tab.

Pwm 4 and 5 only output 1500-2000 when rolling right.
Roll left has no servo movement.

Pitch up will move pwm 4,5, and 6 1500-1000.
Pitch down has no servo movement.

digitalentity, 2017-12-07:
Please post your configuration.

BrianKoch1, 2017-12-08:
# dump

# version
# INAV/OMNIBUSF4PRO 1.7.3 Sep  9 2017 / 10:35:55 (974440df)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature RX_SERIAL
feature MOTOR_STOP
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
beeper DISARMING
beeper ARMING
beeper ARMING_GPS_FIX
beeper BAT_CRIT_LOW
beeper BAT_LOW
beeper GPS_STATUS
[... 458 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2017-12-08:
Please upgrade to 1.8 and verify.

BrianKoch1, 2017-12-08:
Same issue regardless of firmware used.

Same issue on three different boards. Two different planes.

However the odd mixing stops when opening servo tab and checking the servo 4/channel 4 box.

digitalentity, 2017-12-08:
How do you verify that pwm outputs have odd mixing? I used your dump and can't reproduce your issue on 1.8. PWM outputs are correct - verified with a scope.
Please flash with full erase and reconfigure from scratch.

BrianKoch1, 2017-12-08:
I have installed the boards in planes. The servos moving the control surfaces are how i'm checking output.

I'd like to add i've only had this problem with the latest batch of clone boards that have have a green led installed on the board.
Would a youtube video help?

digitalentity, 2017-12-09:
Please answer the following:
1. Does issue persist if you flash with "Full Erase"
2. Is the issue evident in Motors tab?
3. Does the issue manifest itself in stabilized modes?

A video where servos, sticks and "Receiver tab" channels would be visible
might also help.

Also, please provide a CLI dump of a non-working INAV 1.8 setup (after a
full erase and reconfiguring from scratch).

BrianKoch1, 2017-12-09:
1. Yes
2. It is not. Motors tab suggests correct movements are being commanded
3. Yes.

I'm terrible at making videos. For the second half of the video I opened the servos tab and checked the box servo3/channel4 which appears to solve the issue.
https://www.youtube.com/watch?v=3drkLqyiMgI

# version
# INAV/OMNIBUSF4PRO_LEDSTRIPM5 1.8.0 Nov  1 2017 / 06:37:05 (912d1315)

# resources

# mixer
mixer AIRPLANE

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature MOTOR_STOP
feature BLACKBOX
feature AIRMODE
feature PWM_OUTPUT_ENABLE
feature OSD

# beeper
beeper RUNTIME_CALIBRATION
beeper HW_FAILURE
beeper RX_LOST
beeper RX_LOST_LANDING
[... 483 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 20
set pitch_rate = 15
set yaw_rate = 9
set tpa_rate = 0
set tpa_breakpoint = 1500

#

digitalentity, 2017-12-10:
Hm... indeed odd. I still can't reproduce the issue. Servos I connected to my OMNIBUSF4 clone from Banggood work perfectly fine. Can you do this in CLI (just to make sure servos and smix is configured correctly):
```
servo 0 1000 2000 1500 100 -1
servo 1 1000 2000 1500 100 -1
servo 2 1000 2000 1500 100 -1
servo 3 1000 2000 1500 100 -1
servo 4 1000 2000 1500 100 -1
servo 5 1000 2000 1500 100 -1
servo 6 1000 2000 1500 100 -1
servo 7 1000 2000 1500 100 -1
smix reset
smix load AIRPLANE
save
```

Also, can you try this firmware (once again - with full erase and reconfiguring from beginning)? Thanks.
[inav_1.8.1_OMNIBUSF4PRO_LEDSTRIPM5.zip](https://github.com/iNavFlight/inav/files/1545225/inav_1.8.1_OMNIBUSF4PRO_LEDSTRIPM5.zip)

BrianKoch1, 2017-12-10:
Custom mix has the same results.

Firmware linked in previous post has the same results. Full chip erase was selected.

Checking servo3/channel4 box eliminates mixing on both custom smix and linked firmware.

2 boards were bought from a china supplier on ebay and the 3rd was from gear best.

digitalentity, 2017-12-10:
@BrianKoch1 This is really odd. Something is very wrong with the hardware then. May I ask if it's possible to get one of your non-working boards so I can investigate this issue with a scope and a soldering iron? Please email me (address is in my profile) so we can arragne things.

BrianKoch1, 2017-12-11:
Sure. I'll send you two of the them.

DzikuVx, 2017-12-18:
I suppose we can close this one. If anything new will come to the surface, please reopen. Thanks

closed 2017-12-18 by DzikuVx