iNavFlight/inav #2565: Servo weird response when in ANGLE or HORIZON, all fine with PASSTHRU

FinnCloud, 2017-12-07:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_OMNIBUS PRO F4 V3
_INAV/OMNIBUSF4PRO 1.8.0

When in ANGLE or HORIZON the servo move in the right directions but respond very slowly to the command, and servo extension is very limited. When arm (stick arm) the servos' behavior gets even worst, with plane (and board) still and balanced, the servos start to try to correct position and respond very slowly to command. In PASSTHRU servos respond very well no problems. 
What is wrong??
Below is dump.
[dump.txt](https://github.com/iNavFlight/inav/files/1538947/dump.txt)

digitalentity, 2017-12-07:
This is normal. In stabilized modes servos move according to configured rates and PIDs.

FinnCloud, 2017-12-07:
I understand, but why if the plane is flat (the board as well) and still, the stabilize mode try to correct and gets the servo jamming in full extension? I'm expecting form a stabilize mode that when the plane is in correct position the servo are stable and control surfaces are perfectly at center.

digitalentity, 2017-12-07:
Because your plane is not flat. Even 0.1 deg error is not flat for the FC. FC sees this error and tries to correct by deflecting servos by small amount. It doesn't see any response - plane is still not level - so it pushes servos a bit more. Still no response - deflect more etc.

FinnCloud, 2017-12-07:
ok, so this behavior will correct in air when flying?
Just wanted to be sure to do not crush on maiden flight with iNav...
Considering this, can I launch the plane in stabilize mode? or it is suggested to do it in PASSTHRU?

digitalentity, 2017-12-07:
I recommend doing first launch in passthrough if you can fly a plane manually. Then switch the plane into ANGLE in the air and be ready to bail out into PASSTHROUGH if anything goes wrong

FinnCloud, 2017-12-07:
Ok, it makes sense. Thanks for ensuring nothing is wrong with settings regarding stabilized modes.

digitalentity, 2017-12-07:
No worries. Servos moving to their extents on the ground seemingly by their own is a very common concern for most airplane flyers.

digitalentity, 2017-12-07:
I'll close this now.

closed 2017-12-07 by digitalentity