iNavFlight/inav #2568: How to get a old INAV version sotware

MichaelNTJ, 2017-12-08:
I have a board (SP racing F3 )with FW 1.7.2 using INAV configurator 1.7.0 , 
 it was working very well , but after upgrade INAV configurator 1.8.1 ,   the parameter values can not read and  incompatible with my F3 board  ,because I still need to get parameter values from old version ,so

1. I would like to know how to get a old INAV configurator 1.7.0 back for install 
2.anyway can keep old parameters with new FW and nw INAV configurator version?

Thanks

shellixyz, 2017-12-08:
iNav 1.7.2 should work fine with configurator 1.8.1. I just tested it to confirm.

MichaelNTJ, 2017-12-08:
HI, Shellixyz 
could you please also testing FW 1.7.0 too, it will show me error message

shellixyz, 2017-12-08:
It is working fine for me too. What error message do you see ?

MichaelNTJ, 2017-12-08:
the board not here with me , I will upload error message at night

MichaelNTJ, 2017-12-11:
HI, shellixyz  
I am so sorry about I could't get problem message now , because friend update the new FW
so I don't have change testing it again
please close it first till I can find out problem one more time
Thanks.

closed 2017-12-18 by DzikuVx