iNavFlight/inav #2569: UART1 should be off by default on OMNIBUSF4V3

krzysztofmatula, 2017-12-08:
Is there a reason why UART1 is configured to MSP on OMNIBUSF4V3 by default?
(seen on 1.8.0 after flashing with "full chip erase")

This is problematic if there is something (in my case GPS) connected to this port. After flashing with "full chip erase" the configurator is no longer able to connect to the board over USB. The only way to get access to the board is to unplug the GPS. The USB UART is not shared with UART1 on this HW. I think the problem is that the MSP code is overloaded (or hangs) when the GPS keeps transmitting data.

I think the UART1 should just be left unconfigured by default.

digitalentity, 2017-12-08:
UART1 is automatically configured as MSP on boards with native USB to allow connecting to the board if USB port is broken for some reason - we even had a request for this feature.

I don't see how port configured as MSP could be disrupted by connected GPS. The only problem I know about is that the board won't accept DFU connection for flashing if something is connected to UART1, but this is the feature of STM's bootloader and can't be fixed by firmware.

krzysztofmatula, 2017-12-09:
OK, thanks for clarification.
True, DFU is also not working with the GPS on UART1...

Regarding the MSP issue - that's what I observe. The GPS is the NAZA type. Not original, but based on the "pavelsky's" work to interface M8N and NAZA from banggood. On boot-up the GPS module sends some amount of textual debug messages, then switches to binary NAZA protocol. Either of those must cause problems in the MSP parser/handler, as another MSP connection os USB is not working at all.

Anyway, it is a minor issue, as the GPS must be disconnected anyway because of the DFU issue.
Should I close it?

digitalentity, 2017-12-10:
I'm not sure about this. MSP code might be treating "R" character in GPS debug messages as a "reboot to bootloader" command. We might want to add a guard there - the one we already have for entering CLI.

krzysztofmatula, 2019-01-23:
Not sure if we really need to fix this...

closed 2019-01-23 by krzysztofmatula

mirov, 2022-09-03:
Well this is an old thread, but I'm seeing sorta the same thing.  The test case is an F411 board that I've got an ELRS receiver using UART1 with.  If I flash from a newly built image (I'm building my own firmware) it completes, and lets me connect.  It pops up the "pick your type of plane/quad" window, lets me make a choice and then things seem to be working fine.  Unless I go straight into the CLI.  The clue is that it doesn't say "Entering CLI mode", and commands like "status" don't work.  What seems to be happening is that the MSP is by default selected for both USB and UART1 (as discussed in this thread) for a fresh FLASH.  Sometimes <though not always> I get issues and can't "load from file" my backed-up configuration .txt file (it hangs after selecting the file, I never get an "Execute" button).  If I take the minimal step of turning off UART1's MSP selection (to avoid the MSP / serial receiver conflict on UART1) it seems to work just fine.   

So now I know, but this has been causing me grief for quite a while.