iNavFlight/inav #2570: Omnibus F4Pro SBUS not working (Inav Problem?)

MaFel, 2017-12-09:
I´m using the omnibus F4Pro (the one with SD slot) and I can´t get a connection from my FrSky receiver via SBUS to the flight controller.

With Betaflight the same setup is working fine (same serial rx activated) and also Inav is getting connection to the receiver when I´m using SPEKTRUM2048 and move the sticks of my transmitter the bars also moving in Inav and failsafe is off. If I switch back to SBUS, failsafe (at the top on Inav) turns on and all bars are at 1500.

Does anyone know how I can fix the problem?
Already tried to turn on/off the sbus conversation in the cli
To the FC it´s the Omnibus F4 V2.1 from Banggood, Betaflight version 1.8 stable.

digitalentity, 2017-12-09:
What target do you use to flash? Clones often require either `OMNIBUSF4PRO` or `OMNIBUSF4V3` targets. They mostly differ in a way how S.Bus inverter is connected.

MaFel, 2017-12-09:
I'm flashing OMNIBUSF4PRO, all other OMNIBUS___ are not working on the FC. Also Just 1.8 is working, the Versions before aren't.
On betaflight I'm flashing OMNIBUSF4SD wich ist working fine with SBUS.

shellixyz, 2017-12-09:
@MaFel If I understand correctly you said everything is working if instead of choosing S.BUS as receiver type you select SPEKTRUM2048 ?

MaFel, 2017-12-09:
Yes thats right. Of course the Bars are showing wrong values but if I move a stick in the Taranis the Bars are moving in INav (4 at the same time or so)

shellixyz, 2017-12-09:
@MaFel OK so you get something but it is not good. I thought it could be an issue with the configurator but it doesn't look like it. Is your S.BUS receiver on UART3 ?

MaFel, 2017-12-09:
It's on UART6 I checked it with betaflight where SBUS is working. It might be a problem with the board because other buyers on BG are talking about similar issues

shellixyz, 2017-12-09:
Please try this build: [inav_1.8.0_OMNIBUSF4PRO.hex](https://drive.google.com/open?id=1gpJH9ybTJblQgt4m1CE_xtrRiLlpVLbi)

shellixyz, 2017-12-09:
@MaFel You should try the OMNIBUSF4V3 target to see if it works. I think it should work fine and that it is the one should should use.

MaFel, 2017-12-09:
OF4V3 is not working. The board is not connecting to the computer anymore. The only Version that was working is F4Pro and F4Pro Led.
 I dont think there is a second Inverter. I checked with the multimeter the SBus Pin on the FC ist connected to UART6. My GPS ist running in UART3

shellixyz, 2017-12-09:
@MaFel OK thanks. Please reopen the issue as a proper fix has not been found and merged yet.

MaFel, 2017-12-09:
But the .hex you send me works fine!

shellixyz, 2017-12-09:
Yes but I don't know if we need to add another target or what to allow the fix to be made in the source code for everybody. For the moment people using the configurator can't flash a working version directly.

shellixyz, 2017-12-09:
@digitalentity This looks like a bug. It looks like `OMNIBUSF4PRO` or `OMNIBUSF4V3` can't be used on this board. Should a new target be created ? The fix consisted in adding the inverter line for UART6 in the `OMNIBUSF4PRO` target.

MaFel, 2017-12-09:
ok, here is a corrected pinout from the comments. 
The resistance between the the SBUS-pin UART6 0.7ohm so the might be a inverter inbetween.

[image]

shellixyz, 2017-12-09:
@MaFel I doubt it because 0.7ohm is a dead short. It is probably what you get when touching the probes of your multimeter together.

digitalentity, 2017-12-10:
@shellixyz what did you change to buid that hex?

shellixyz, 2017-12-10:
@digitalentity I did this https://github.com/shellixyz/inav/commit/31d1aa94e2792398821af6c18910bbd17d5d8c79

digitalentity, 2017-12-10:
Aha, so there is another option for SBUS - UART3. Crap, this is yet another board variant.

digitalentity, 2017-12-10:
Yes, we probably need another target for this - with removed I2C (which is shared with UART3 and is unusable on that board due to inverter).

shellixyz, 2017-12-10:
UART6 was the problem. @MaFel said his receiver is connected on UART6. That's why I advised him to try the OBV3 target but for some reason it didn't work. Only OBPRO with UART6 inverter enabled worked.

shellixyz, 2017-12-10:
I'm not sure about UART3. @MaFel Did you intentionally wrote UART6 in your [comment](https://github.com/iNavFlight/inav/issues/2570#issuecomment-350510793) or were you talking about UART3 ?

digitalentity, 2017-12-10:
Odd. OMNIBUSF4V3 should have worked then. Unless the board has MPU6500 instead of MPU6000. @MaFel can you post output of CLI `status` command?

digitalentity, 2017-12-10:
The fact that OBV3 target was unable to connect to configurator probably indicates that there's a gyro detection issue - OBPRO allows both 6000 and 6500 gyros, while OBV3 only has 6000 driver enabled. I think we can adjust the target to enable 6500 driver for OBV3 as well.

shellixyz, 2017-12-10:
Agreed

MaFel, 2017-12-10:
# status
System Uptime: 44 seconds
Current Time: 2017-12-10T10:38:41.609+00:00
Voltage: 1 * 0.1V (1S battery - NOT PRESENT)
CPU Clock=168MHz, **GYRO=MPU6500, ACC=MPU6500**, BARO=BMP280, MAG=HMC5883
STM32 system clocks:
  SYSCLK = 168 MHz
  HCLK   = 168 MHz
  PCLK1  = 42 MHz
  PCLK2  = 84 MHz
Sensor status: GYRO=OK, ACC=OK, MAG=OK, BARO=OK, RANGEFINDER=NONE, OPFLOW=NONE, GPS=OK
SD card: None inserted
Stack size: 6144, Stack address: 0x10010000
I2C Errors: 0, config size: 3333, max available config: 131072
ADC channel usage:
   BATTERY : configured = ADC 2, used = ADC 2
      RSSI : configured = ADC 3, used = none
   CURRENT : configured = ADC 1, used = none
  AIRSPEED : configured = none, used = none
System load: 1, cycle time: 2002, PID rate: 499, RX rate: 49, System rate: 9

# 

UART1 is GPS
UART3 is Compass
UART6 is SBUS teceiver

lancaster100, 2017-12-10:
Mafel
I have exactly the same problem with the Omnibus F4 Pro v2.1 board using T9Xtreme with xjt module using Ersky9x and the X8R
How did you apply the Hex file to get the sbus working
regards Hans

MaFel, 2017-12-10:
Download the Hex File above and click local File in the Firmware flash tab

shellixyz, 2017-12-10:
@MaFel Could you please try this: [inav_1.8.0_OMNIBUSF4V3.hex](https://drive.google.com/open?id=14_oRzBfa29iXhyvaAvwGK1078l85pkvF)
This should be the proper fix for your FC.

lancaster100, 2017-12-10:
Mafel
Thanks for the quick reply
I have INAV v1.8 already flashed  - do I flash the local hex file after this?
regards
Hans

MaFel, 2017-12-10:
@shellixyz the the firmware works fine if I recognize any problems I will let you know. Thanks!
Is the issue ready to be closed? :D

@lancaster100 yes, if the firmware before is not working flash the one shellixyz uploaded.

shellixyz, 2017-12-10:
@MaFel Great ! When the next version of iNav gets out you should flash OMNIBUSF4V3 and it should work with the build downloaded directly with the configurator. Yes you can close this issue I will create a pull request to replace it :)

closed 2017-12-10 by MaFel

lancaster100, 2017-12-10:
Mafel
Hi
Flashed the local hex file ok
I also repeated the set up settings in INAV just in case - but still no response on output bars in the receiver tab using SBUS and X8R ( all connected and bound ok)
I notice on Matts video he has set the mode switches on the Taranis at this point 
I am using 9XEtreme and Ersky9X so my switch setup is different
I have simple channel settings on AETR but do the extra channel settings affect SBUS response?
Any ideas?

xpepe4, 2017-12-15:
Hi there, i have the same problem with the same FC.
I tried to flash with shellixyz fixes, and now sbus work well on UART6.
The problem is the GPS who's now in conflict on the same UART ! GPS is not working on others UART !
What can i do ? please help !!!

shellixyz, 2017-12-15:
@xpepe4 The create a new issue with setup details and config dump. You can only have one device connected on each UART.

xpepe4, 2017-12-15:
Reading an other post i understand my error... i just have to rewire the gps to uart3 and it works fine now.
Thanks for the help ! :)

Rijaylove, 2018-06-14:
Hello I have the F4 stack/VTX combo and I bind it to the receiver it’s green but in Betaflight no response on the receiver tab.  I updated the firmware to the new 3.3 version still no Luck. I read the instructions on how to bridge the pad 1 so I can use Sbus still no luck.

modellbobby, 2018-08-09:
Hello, I have the exact same problem with the Board 

I flashed the OMNIBUSF4PRO target

Omnibus F4 Pro V2.1 from bang good. Frsky receiver on UART6 does not work. 

What target should I use now?

lancaster100, 2018-08-11:
Modelbobby
Hi, I used the Omnibus F4 Pro V2.1 from Banggood
I flashed it with  inav_1.8.1_OMNIBUSF4V3.hex
<https://drive.google.com/open?id=1M4oHqC0AWE8ZF_hzX2UWD8Sv2wyuhJfk> ,   target
OBF4V3 and connected my FRsky X6R SBUS to UART6
GPS  then goes to UART3
hope this helps
Hans

FXvBH, 2020-05-03:
Hello. Hopefully I will join this discussion in the right way.

I Have this FC: https://www.banggood.com/Eachine-Wizard-X220S-FPV-Racer-RC-Drone-Customized-Omnibus-F4-V3-S-Betaflight-3_0_1-Flight-Controller-p-1177121 .html? rmmds = myorder & cur_warehouse = CN

I have this GPS: https://www.banggood.com/Flight-Controller-GPS-with-Protective-Shell-for-PIX-PX4-Pixhawk-p-1005394.html?rmmds=myorder&cur_warehouse=CN
I am using INAV 2.4.0

I have for TX / RX FrSKy Taranis X9D plus / Receiver X8R

On UART6 with only GPS the GPS is working
On UART6 with only receiver X8R the receiver is working

But, by using both there is a problem. The GPS or Receiver are not working on UART1 or UART3. By using both there is a conflict.

hopefully you have advice

Greetz, FX from the Netherlands.

FXvBH, 2020-05-03:
Gentlemen
I have transferred the GPS from pinout RX6 / TX6 to RX1 / TX1, and set the GPS to UART 1, and now it seems to be working. The GPS changes from red to blue, Total messages increases, and the receiver simply works on AURT6. Would it still be possible?

lancaster100, 2020-05-04:
Hi
My flight controller was Omnibus F4 V2.1
UART connections were as follows:-
UART 1    GPS
UART 3    COMPASS
UART 6    SBUS RX
 Good luck
Hans

rakiata, 2020-08-03:
Hi, I used the Omnibus F4 Pro V3 from Aliepress
I flashed it with  INAV/OMNIBUSF4PRO_LEDSTRIPM5 2.5.0
UART connections:
UART 1 -  SBUS RX / UART 3 - COMPASS /  UART 6 - GPS
After calibrating all sensors, I put the battery and the gyro in the sensor tab sits in a straight line when I move it. I went outside to find satellites and everything was fine until I arm it. Navigation is safe but the gyro does not want to work. I hope it helps with some advice

[dump 2.5.txt](https://github.com/iNavFlight/inav/files/5017211/dump.2.5.txt)
[status 2.5.txt](https://github.com/iNavFlight/inav/files/5017212/status.2.5.txt)

RCfavorite, 2022-10-30:
Sorry open discussion again
I have omnibus f4 AIO i flash target f4 omnibus. Sbus not working on any target. Can any body help me?