iNavFlight/inav #2571: Any chance to get Omnibus F4 Channel Forwarding working? [support]

parnz, 2017-12-09:
I've this board and I don't use LED.
Needed this feature to forward aux to PWM5 and PWM6 to control gimbal
My setup is QuardX mainly use for take video. 

[image]
[image]
[image]

I read some where that because timer conflict.
I know coding as I'm a programmer but in the MCU world don't know how pin and timer work.
I do coding in Arduino but there is no timer and pin related.
 
 Any suggestions are all appreciated.
Thanks

chan2328, 2017-12-09:
same matter #2434, at this moment, inav can not support to change output pin for the mixer QuardX as fixed output pin, I just know that it is available to change output pin by the CLI command "RESOURCES" in betaflight. you can study it https://github.com/betaflight/betaflight/wiki/Betaflight-resource-remapping or may be you can study #2328 to do coding as well.

parnz, 2017-12-09:
afaik inav does not support command RESOURCE mapping.

But I'm looking into another way - use dedicated hardware (Arduino) to extract IBUS to PWM externally.
[image]

parnz, 2017-12-11:
I managed to get it working by adding arduino to decode IBUS to PWM.
I can control gimbal by AUX channel now.
Plus I can detect a failesafe signal which I assigned to channel10.
When failesafe activated it will re-center the camera and return to home.

closed 2017-12-11 by parnz

giacomo892, 2017-12-11:
Do you mind sharing how you detect failsafe on iBUS?

flybabo, 2018-01-31:
If you use a FrSky X4RSB receiver, you can access 2 channels (CH9,10) in addition to the SBUS and CPPM.
I'm using those 2 channels for pan and tilt servos of video cam without going through the FC.
Note that if you program the receiver with non-CPPM firmware, you can access 3 channels (CH9,10,11).