iNavFlight/inav #2572: Betian BN 880  GPS not receiving any satellites

BVN99, 2017-12-09:
### My GPS seems to be connected but doesn't receive satellites :(
[image]
 any Sats :(

I'm using a  Flip 32 f4 v3pro {Omnibus} with barometer built in, and despite GPS being connected, I receive no sats. It perpetually waits for 3D GPS fix {PFA the image}. I've tried waiting and moving outside but to no avail. It's running Inav 1.8.0. Thanks in advance!

shellixyz, 2017-12-09:
@BVN99 To help confirm it is not a hardware problem you could try with Cleanflight or Betaflight

KiwiDavid, 2017-12-09:
Try to keep it above and away from any other electronic devices.  I found both the FC (SPRacing F3) and my receiver (FS-ia6B) produced considerable in-band noise that swamped the GPS.

digitalentity, 2017-12-10:
Give it a while. Some GPS units can take up to 5-10 minutes to acquire a fix. There are no errors meaning GPS is talking to INAV fine, which gives a hint that it might be a hardware issue.

DzikuVx, 2017-12-11:
This looks like a GPS hardware failure or just poor conditions or a really cold GPS start. I'm closing this one. If you have are new information indication this might be a INAV problem, please report back

closed 2017-12-11 by DzikuVx

BVN99, 2017-12-11:
@digitalentity  Thanks so much, this actually did help. Expected to see it work in a mnute but took 4 to get a fix. All good now! :D

romko1, 2018-12-05:
@digitalentity: After updating FW to INAV 2.0.1, it takes a long time to get a GPS 3D fix (cold start 5-8min).
 I have a MATEKF405 -> UART5 for GPS, GPS module: Beitian BN-880. When I had FW 1.9.1 or 2.0.0 so the cold start was shorter.
 I have a BAUT-rate of 57600. When connecting GPS to u-center, I have a 3D fix within 2 minutes indoor.
 Did anyone have such a problem? 
 Thank you.

laurentopia, 2019-02-08:
@romko1 some BN-880 come with faulty battery and it seems that this GPS uses volatile memory to store satellite lock. I suggest disconnecting it and checking batt voltage, it's the tiny button battery under the GPS antenna.