iNavFlight/inav #2574: OSD stats won't fit NTSC screen

krzysztofmatula, 2017-12-10:
For some reason I use NTSC for FPV and the final flight stats won't fit the screen:
[image]

There should be one more line - "DISARMED BY", which is not visible.
Maybe the whole stats could be printed one line up, as there is space on the trop?

It would be also nice to have odometer stats on this screen (total flight time and total distance), but how to fit them?

shellixyz, 2017-12-10:
This could be because of the settings of your DVR. With the Eachine ProDVR I need to select the DL option to see all the lines of the OSD. The goggles can crop the image also. Is it the same in your goggles ? But yes the stats could probably be moved one line up.

Aren't total flight time and total distance already on this screen (traveled distance and fly time) ?

krzysztofmatula, 2017-12-10:
By odometer stats I mean `stats_total_time` and `stats_total_dist` (cumulative stats, not the last flight).

digitalentity, 2018-02-06:
Fix merged. Closing

PR #2721 (2018-02-06): OSD stats improved
- stats now fit NTSC screen
- reordered items - altitude close to distance
- odometer displayed on welcome screen (stats screen is full)

Fixes #2574
commit 3aa0d10a (2018-02-06): OSD stats improved

- stats now fit NTSC screen
- reordered items - altitude close to distance
- odometer displayed on wellcome screen (stats screen is full)

Fixes #2574
closed 2018-02-06 by digitalentity