iNavFlight/inav #2580: 1.8 Omnibus F3 Pro, Servo Configuration Issue

axcodeplex, 2017-12-12:
### Board and Version
Omnibus F3 Pro v2
v1.8

### Behavior
* _Describe the problem_
Configured with flying wing preset, servos connected, motor connected. Mix works, craft flies as expected, servos work, all fine.
However, manual changes to the servos' config, either via the cli or the gui, have no effect (direction, centerpoint, etc). The changes stick past reboot and are reflected if queried in the cli or gui, but have no impact

* _Steps needed to reproduce the problem_
Change any active servo parameter, observe if changes are reflected in reality of servo operation.

* _Expected Results_
Example: Change of servo midpoint, 1500 to 1600. Expected, servo should alter standard midpoint

* _Actual Results_
Servo shows no change in midpoint (tested with midpoint and direction, on both servos, via both cli and gui)

_Is this a new feature request?_
No

#### Blackbox logs/Config (.ZIP) where problem occurs
_Go to CLI, execute `dump` command copy its output to [PasteBin](https://pastebin.com) and provide a link to a paste here.
Dump forthcoming (need to generate one)

_Upload a zipped blackbox log (if available)_
Blackbox not applicable.

shellixyz, 2017-12-12:
I think your not changing the parameters of the right servos. For a flying wing elevon servos are servos 2 and 3.

axcodeplex, 2017-12-12:
@shellixyz I'll validate by trying the other servos, but should point out that my dump above does not have the attempts at set present. Thanks for the input, I'll give it a crack.

axcodeplex, 2017-12-12:
HA. Well... Interesting. Turns out servos are on 3 and 4, not 0 and 1, despite 0 and 1 being listed as channels 3 and 4?
https://www.dropbox.com/s/m6scfleybv72dap/2017-12-12%2015_22_45-INAV%20-%20Configurator.png?dl=0

Either way, clearly my own stupidity, not a bug - closing out now. Thanks for the dose of common sense @shellixyz !

closed 2017-12-12 by axcodeplex

shellixyz, 2017-12-12:
@axcodeplex  It is just that ouputs are numbered from 1 but servos are numbered from 0. So output 3 = servo 2.