iNavFlight/inav #2581: Feature Request: PWM1 for servo

janbolt, 2017-12-13:
It would be great if PWM1 could be used as servo output.
Most airplanes only have one motor but possibly many servos.

shellixyz, 2017-12-13:
There is PR in progress discussing this: #2328

janbolt, 2017-12-13:
Not sure if that does what I want.
I stumbled over Mixer.md which says
"Currently, pin id's 0 and 1 can only be used for motor outputs. "

shellixyz, 2017-12-13:
Yes currently. If you want to use PWM1 for a servo then what is discussed is what will enable this possibility through dynamic allocation of the outputs opposed to static allocation as it is now allowing only max 2 motors and the two first outputs reserved for motor use.

jflyper, 2017-12-14:
@janbolt It depends on a board and a timer assignments if you can actually use only one motor output, and even if it's possible, it may not be the first output pin. This is a limitation of imposed by MCU hardware.

#2328 will allow user to dynamically assign motors and servos within the limitation.