iNavFlight/inav #2582: I2C OLED SSD1306 not working with ext HMS5883

pipersw, 2017-12-13:
### If your issue looks like a hardware fault or a configuration problem please don't raise an issue here.

### Please consider using other user support options such as asking the manufacturer of the hardware you are using, [RCGroups](https://www.rcgroups.com/forums/showthread.php?t=2495732), [Slack](https://inavflight.signup.team) or other user support forums & groups (e.g. facebook).

### Board and Version
_Board name goes here. If you have a link where you've purchased the board - please include it_
Revolution F4

_Version of INAV used goes here_
INAV 1.8.1

_Use CLI command `version` and put its output here_
INAV/REVO 1.8.0 Nov  1 2017 / 06:39:22 (912d1315)

# resources

# mixer

# servo

# servo mix

# feature
feature -BLACKBOX
feature GPS
feature DASHBOARD
feature PWM_OUTPUT_ENABLE

# beeper

# map

# name

# serial
serial 0 2 115200 115200 0 115200
serial 5 1 115200 38400 0 115200

# led

# color

# mode_color

# aux
aux 0 1 0 900 1200
aux 1 4 0 1300 1700
aux 2 3 0 1300 1700

# adjrange

# rxrange

# master
set looptime = 1000
set gyro_sync = ON
set gyro_sync_denom = 1
set gyro_hardware_lpf = 188HZ
set gyro_lpf_hz = 70
set gyro_notch1_hz = 170
set gyro_notch1_cutoff = 125
set gyro_notch2_hz = 85
set gyro_notch2_cutoff = 43
set acc_hardware = MPU6000
set acczero_x = 137
set acczero_y = -12
set acczero_z = -599
set accgain_x = 4095
set accgain_y = 4103
set accgain_z = 3986
[... 549 lines cut ...]
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 65
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 180
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 10
set pitch_rate = 10
set yaw_rate = 10
set tpa_rate = 0
set tpa_breakpoint = 1500

_Upload a zipped blackbox log (if available)_

pipersw, 2017-12-13:
Multiple devices on I2C  are allowed ?

pipersw, 2017-12-14:
Test done with i2c_speed set at 400KHz, 200KHz and 100KHz.
I2C address in INAV code is 0x3C, which correspond to 0x78 on the screen (0x3C *2).

shellixyz, 2017-12-14:
Is the screen working when alone on the bus ? Is the compass working when alone on the bus ?

pipersw, 2017-12-14:
Compass is working when alone on the bus, at default speed 400KHz.
I have to test the screen working alone on the bus.
the resolution of my screen is 64x48, it is a problem ?

pipersw, 2017-12-14:
In file display_ug2864hsweg01.h I see 
#define SCREEN_WIDTH 128
#define SCREEN_HEIGHT 64

So my screen size (64x48) is not compatible. I close the issue.

pipersw, 2017-12-21:
Test done with 128x64 OLED : no display with INAV, with the screen alone on the CC3D.
Screen is working on Arduino Uno with lib from Sparkfun : [https://github.com/sparkfun/SparkFun_Micro_OLED_Arduino_Library](url)

DzikuVx, 2018-05-09:
lowering `i2c_speed` helps sometimes. Also we know that ssd1306 works so I suspect this is hardware specific issue. Because so, I'm closing this one. If new data appears, feel free to reopen

closed 2018-05-09 by DzikuVx