iNavFlight/inav #2586: `RANGEFINDER_HCSR04_PWM_TRIGGER_PIN`?

jflyper, 2017-12-14:
I stumbled across this one while porting the rangefinder abstraction to BF.

I see `RANGEFINDER_HCSR04_PWM_TRIGGER_PIN` on the first conditional at line 74.
I don't fully understand what these conditionals are doing, but I have a strong feeling that this is a typo for `RANGEFINDER_HCSR04_TRIGGER_PIN_PWM`.

https://github.com/iNavFlight/inav/blob/master/src/main/sensors/rangefinder.c#L74-L92
```
 74 #if defined(RANGEFINDER_HCSR04_PWM_TRIGGER_PIN)
 75     // If we are using softserial, parallel PWM or ADC current sensor, then use motor     pins for sonar, otherwise use RC pins
 76     if (feature(FEATURE_SOFTSERIAL)
 77             || (rxConfig()->receiverType == RX_TYPE_PWM)
 78             || (feature(FEATURE_CURRENT_METER) && batteryConfig()->currentMeterType ==     CURRENT_SENSOR_ADC)) {
 79         rangefinderHardwarePins.triggerTag = IO_TAG(RANGEFINDER_HCSR04_TRIGGER_PIN_PWM    );  
 80         rangefinderHardwarePins.echoTag = IO_TAG(RANGEFINDER_HCSR04_ECHO_PIN_PWM);
 81     } else {
 82         rangefinderHardwarePins.triggerTag = IO_TAG(RANGEFINDER_HCSR04_TRIGGER_PIN);
 83         rangefinderHardwarePins.echoTag = IO_TAG(RANGEFINDER_HCSR04_ECHO_PIN);
 84     }
 85 #elif defined(RANGEFINDER_HCSR04_TRIGGER_PIN)
 86     rangefinderHardwarePins.triggerTag = IO_TAG(RANGEFINDER_HCSR04_TRIGGER_PIN);
 87     rangefinderHardwarePins.echoTag = IO_TAG(RANGEFINDER_HCSR04_ECHO_PIN);
 88 #else 
 89     // No Trig/Echo hardware rangefinder
 90     rangefinderHardwarePins.triggerTag = IO_TAG(NONE);
 91     rangefinderHardwarePins.echoTag = IO_TAG(NONE);
 92 #endif 
```

digitalentity, 2017-12-14:
Good spot. I'll check if this is needed at all

digitalentity, 2018-02-13:
I'm not sure this needs fixing. It's a leftover from F1 firmwares that are not supported anyway. We need to consider cleaning up the F1 leftovers.

digitalentity, 2018-02-25:
Moving to 2.0. We'll clean this mess up when removing F1 code.

DzikuVx, 2018-04-25:
removed in #3121 closing

closed 2018-04-25 by DzikuVx