iNavFlight/inav #2587: feature GPS broken in 1.8.1 dev

stronnag, 2017-12-14:
With 7e280defb64 (and possibly a few earlier), `feature GPS` in the CLI is broken.
````
# serial
serial 1 2 115200 38400 0 115200

# feature
Enabled: VBAT BLACKBOX 

# feature GPS
unavailable
````
Enable GPS from the configurator.
````
# feature
Enabled: VBAT GPS BLACKBOX 
````
I'm pretty sure this got me a week ago when the GPS failed to restore from a `diff`.

shellixyz, 2017-12-14:
Bug is in 7a1491. Fixed in PR #2590

stronnag, 2017-12-14:
Thanks, I could have grepped for 'unavailable' had I put my mind to it. I wonder how many more of these missing USE_ cases we have?

shellixyz, 2017-12-14:
I'll try to check to see if others could have been missed

stronnag, 2017-12-14:
#2590 fixes this on SPRF3 but not on OMNIBUS ??? What?

shellixyz, 2017-12-14:
I can't see how it is possible :grin:

stronnag, 2017-12-14:
PEBCAK

closed 2017-12-14 by stronnag

shellixyz, 2017-12-14:
I found some more. I opened another PR #2591