iNavFlight/inav #2588: Latest MATEK F405 changes prevent legacy F405OSD from booting.

stronnag, 2017-12-14:
The changes to support the new Matek F405 board variants prevent the legacy F405OSD from booting.
 7e280defb642d85f45cfbb784a57b6e051935400 will not boot.

reverting to 9a3cf280 allows the board to boot again

digitalentity, 2017-12-15:
Will investigate, thanks

digitalentity, 2017-12-15:
F405OSD board (not AIO). Boots fine. @stronnag please try flashing with full erase.
Maybe it's some specific config that prevents the board from booting?

stronnag, 2017-12-15:
It was  flashed with full erase, twice, `dfu-util` with the `mass-erase` option. I'll try the configurator later and post the diff.

digitalentity, 2017-12-15:
My bad, checked the wrong commit. Indeed it doesn't boot. Strange, I definitely checked that commit before merging.

stronnag, 2017-12-15:
I did the same thing yesterday on the use_gps fix test. :)

digitalentity, 2017-12-15:
Damn. Found it! I left a `while(1);` in MPU6000 detection code ("not detected" execution path). Used that when debugging `busDevice` code. Matek board doesn't have the MPU6000 so it got stuck in infinite loop.

stronnag, 2017-12-15:
That might explain another strange non booting migrate to busdev ...

digitalentity, 2017-12-15:
Fix ready. Closing. @stronnag thanks for reporting!

PR #2593 (2017-12-15): Remove rogue empty loop in mpu6000 detection code
Fixes #2588
closed 2017-12-15 by digitalentity