iNavFlight/inav #2589: Omnibus F4Pro v2.1 SBUS response - INAV problem persists

trivejb, 2017-12-14:
I seem to hit the same problem as lancaster100 in a previous post (now closed) in getting my X8R and GPS to connect to Flip32 F4V3 Omnibus( from RTFQ).
This is the board I have;  http://www.readytoflyquads.com/flip-32-f4-omnibus-v3

 I have tried all the different versions of the firmware (including the one mentioned here) in addition to going back to the BataFlight firmware listed on the RTFQ website. 

 The receiver is bound and I have verified that the SBUS is working along with the PWM outputs (which of course I do not use).

I have tried turning on ports, some will reset back to off after I do the Update and Reboot sequence and sometimes the channel bars will disappear in the radio section.  I then have to close the CLI and start again.

 If I set the receiver mode in the tools tab to SERIAL-BUS/ BUS and then come back it will be set to off again.

I have verified all headers are adequately soldered and nothing is bridged and all wiring plugged in correctly.

The camera in and video work fine as well as the OSD.

I have removed all accessories (GPS, Camera, VTX) from the board but does not help.

Feel like I have overlooked something simple but am totally stumped.

Any suggestions would be appreciated.

shellixyz, 2017-12-14:
What firmware and configurator version are you using ?

trivejb, 2017-12-14:
Using configurator 1.8.0 Windos version. Have also used chrome version 1.8.1
Firmware as uploaded by configurator 1.8 Omnibus F4V3 2017-1-1 10:13  plus the version suggested in the lancaster100 thread.
also went back to BataFlight with same issues
I can turn on Uart6 and it sticks but receiver source always goes back to 'NONE"

shellixyz, 2017-12-14:
Try to set the options manually in the CLI:
```
set receiver_type = SERIAL
serialrx_provider = SBUS
save
```

Be sure to have port configured as RX and check if your receiver is working in the receiver tab.

trivejb, 2017-12-14:
serialrx_provider = SBUS was in invalid comand and I could not find anything similiar

shellixyz, 2017-12-14:
Try again with `set serialrx_provider = SBUS` I forgot to put the set in front.

Also you are saying you have the same issues with Betaflight. Can I ask which issues ? The saving problem, the RX problem, both ?

trivejb, 2017-12-14:
Yes, the exact same problems with Betaflight.   set serialrx_provider = SBUS worked this time but did not change anything

shellixyz, 2017-12-14:
Where did you connect your receiver on your FC ?

If you have issues saving the config with iNav and also Betaflight it is probable it is an hardware issue with your FC. Is your FC new ?

trivejb, 2017-12-14:
Yes, Just got if from RTFQ. I was also thinking it was a bad FC but because I'm just learning this wanted to make sure it was not something I was overlooking.  I am going to send an email to RTFQ and hope for a response as they do not have any contact methods listed on their web site. I have already spent a number  of hours on this and about to give up. Would be willing to send to someone if they wanted to mess with it and pay shipping back.
Thanks,

shellixyz, 2017-12-15:
What was the firmware on your FC when you received it from RTFQ ?

Last thing I can advise you to do is try "full chip erase" in the configurator if you didn't do it before and to use only the firmwares from the configurator and matching your target (OMNIBUSF4V3). If you are flashing the wrong target you can mess up your FC.

The builds I posted in the issue you mentioned did not solve the problem because it was not coming from the software and does not apply to your board anyway, yours is different it is not the Pro version.

What country are you living in ?

trivejb, 2017-12-15:
OK... found one problem... my fault! Missed a pin when plugging in the receiver to the board. Now my receiver type and port UART6 sticks but still not any movements of the bars on the receiver tab but at least I'm making progress.
USA, Tennessee

shellixyz, 2017-12-15:
The thing is when you enable serial RX in the ports tabs you don't need anything connected for it to stick. Even when nothing is connected it should save fine. `receiver_type` and `serialrx_provider` also.

trivejb, 2017-12-15:
going to try another laptop??? about the only thing I have not tried

shellixyz, 2017-12-15:
Can't hurt.

shellixyz, 2017-12-15:
Please change the title to better reflect the issue "RX issues with Flip32 Omnibus F4V3 from RTFQ" for example would be more appropriate.

trivejb, 2017-12-16:
OK... NEWBIE problem, as I knew it would be. NOT A PROBLEM WITH THE BOARD!  When trying to make UART 6 active I did not understand what "MSP" was (and still do not) and was trying to turn it on. I then found a listing on Banggood showing a screen print of the port settings and have now only set Serial RX on and receiver to SBUS and now have the receiver working. Moved GPS connection to UART 1 and it works. I think that RTFQ had RX/TX marked in reverse.  Mag was staying RED and not sure why but should be able to figure it out.
Thanks for all the input and sorry for the confusion

shellixyz, 2017-12-16:
Great ! I should have asked for your config dump. MSP (Multiwii serial protocol) is the protocol you are using through the configurator to query/set data/config of the board. It can be used on a UART to connect an OSD or to enable configuration through a Bluetooth adapter for example. You should only enable one thing on each UART, with some exceptions.

shellixyz, 2017-12-16:
@trivejb Can this issue be closed ? Consider joining the [slack channel](https://inavflight.slack.com) you can ask questions there.

trivejb, 2017-12-16:
Yes, just did. thanks for bringing this to my attention

closed 2017-12-16 by trivejb

trivejb, 2017-12-16:
Very interested but If I understand it correctly I need an invitation to join the inavflight workspace. Is this correct or how do I go about it?

shellixyz, 2017-12-17:
@trivejb It should be open for everybody. I for sure didn't need an invitation. Try this https://slack.com/get-started#join