iNavFlight/inav #2592: only 1 motor in airplane mode

wx4cb, 2017-12-15:
Latest version, currently working with @shellixyz ont he betaflight F4 target. Question.

I have a regular fixed wing plane which only has 1 motor. The BFF4 board only has 4 esc/servo outputs.

How do I tell it that it only has 1 motor, but ail/rud and ele on channels 2/3/4 ?

I'm assuming i have to do a custom mix?

shellixyz, 2017-12-15:
For the moment you can't without customizing the code

This issue is a duplicate of #2581

shellixyz, 2017-12-15:
@wx4cb I don't really have the time to look now but if you want to give it a go motor/servo pwm mapping is done in drivers/pwm_mappings.c function pwmInit line 60. It shouldn't be too complicated to disable the second motor output and shift the servo outputs.

shellixyz, 2017-12-15:
@wx4cb Be aware that because PWM1 and PWM2 share the same timer on the BFF4 FC you won't be able to use a different protocol for ESC/servos if you do this mod. Using standard 50Hz PWM is advised.

wx4cb, 2017-12-15:
that's all I would be using anyway is the standard PWM - it is a regular plane ESC not a fancy DSHOT one like on my quads :dancing_men:

wx4cb, 2017-12-15:
is the R7 code that you just did with the i2c patches available here still: https://github.com/shellixyz/inav/tree/BETAFLIGHTF4_target ?

shellixyz, 2017-12-15:
Yes it is you can see the last commits here: https://github.com/shellixyz/inav/commits/BETAFLIGHTF4_target

shellixyz, 2017-12-16:
@wx4cb I did not test but maybe you only need to replace TIM_USE_FW_MOTOR by TIM_USE_FW_SERVO for output 2 in the target.c file

wx4cb, 2017-12-16:
ok i'll take a look... just got home from shopping

wx4cb, 2017-12-16:
@shellixyz  Does this look right? I'm assuming that they're in "motor order"

    // Motors
    { TIM3, IO_TAG(PB0),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR | TIM_USE_FW_MOTOR }, // S1_OUT D1_ST7
    { TIM3, IO_TAG(PB1),  TIM_Channel_4, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM3, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO }, // S2_OUT D1_ST2
    { TIM8, IO_TAG(PC9),  TIM_Channel_4, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO }, // S3_OUT D1_ST6
    { TIM8, IO_TAG(PC8),  TIM_Channel_3, 1, IOCFG_AF_PP_PD, GPIO_AF_TIM8, TIM_USE_MC_MOTOR | TIM_USE_FW_SERVO } // S4_OUT D1_ST1

shellixyz, 2017-12-16:
Yes

wx4cb, 2017-12-16:
ok so now correct me if i'm wrong, I can just change the "custom aurplane" mixer in mixer.c to below and that should give me 1 motor output and 4 servo outputs ?

From: 
DEF_MIXER( MIXER_CUSTOM_AIRPLANE,   PLATFORM_AIRPLANE,   2,          true,       true,       NULL ),

To:
DEF_MIXER( MIXER_CUSTOM_AIRPLANE,   PLATFORM_AIRPLANE,   2,          true,       true,       NULL ),

shellixyz, 2017-12-16:
I don't know but you can try. Worse case scenario it won't work. it doesn't work without changing that ?

shellixyz, 2017-12-16:
If you can't make it work I'll have a look tomorrow

wx4cb, 2017-12-16:
yea that was a typo... I tried that, and also tried enabling the CUSTOM_AIRPLANE mix with 1 motor... but I can't quite figure out how it assigns servos to channels.
I would have assumed that with a single motor defined in the mix, and use_servos turned on, then channel 2 would be servo 0 maybe? 

But I'm not getting anything showing up with movement in the motors tab like I do when the mixer_airplane is setup (all channels move as per the drawing in the bars on the right".

I'm trying to figure this out, but maybe i'm just not understanding how it uses the servos internally?

wx4cb, 2017-12-16:
In mixer.c i did this:

// Single motor Mix Map
static const motorMixer_t mixerSingleProp[] = {
    { 1.0f,  0.0f,  0.0f, 0.0f },
};

then int he array assignment:
//DEF_MIXER( MIXER_CUSTOM_AIRPLANE,   PLATFORM_AIRPLANE,   2,          true,       true,       NULL ),
// Create custom airplane mix with just 1 motor output
    DEF_MIXER( MIXER_CUSTOM_AIRPLANE,   PLATFORM_AIRPLANE,   1,          true,       true,       mixerSingleProp ),

I also re-enabled the helicopter mixes thinking that might be another option that I can copy, but atm i can't figure out how it assigns servos in  the mixer

wx4cb, 2017-12-16:
one thing I did notice though, the mixer for flying wings is wrong too.

the image shows motor on 1/2 and servos on 3/4, however when i selected that, it was showing on 5/6 on the motors tab? maybe that's a clue?

shellixyz, 2017-12-16:
This part of the code is still obscure to me. Understanding the big pwmInit function is needed to know how motors/servos are mapped to the outputs. I think it uses the number of motors in the mixer definition. It begins by reserving the motor outputs then puts the  servo outputs on the next free outputs.

wx4cb, 2017-12-16:
that's logical, however, unless im misunderstanding something then

    DEF_MIXER( MIXER_AIRPLANE,          PLATFORM_AIRPLANE,   1,          true,       true,       mixerSingleProp ),

should make motor1 output to be the motor and 2/3/4 to be servos and the other channels. however, it still doesn't map the servos down to there, they stay at 3/4/5/6basically what I need is servo 6 to move down to the #2 position so it would be 2/3/4/5

shellixyz, 2017-12-16:
In pwm_mapping.h you can try change

> #define MAX_MOTORS  4

to

> #define MAX_MOTORS  1

shellixyz, 2017-12-16:
I did a quick test and just replacing TIM_USE_FW_MOTOR by TIM_USE_FW_SERVO for output 2 in the target.c file is working for me. After this change the first output is mapped to the motor and the rest are mapped to servos.

shellixyz, 2017-12-16:
@wx4cb try this: [inav_1.8.1_BETAFLIGHTF4_one_motor.hex](https://drive.google.com/open?id=13xn2_XjbDoosTjapQrxA3TW6wbvOw0Lr)

wx4cb, 2017-12-16:
yea that's as far as I got. the issue now is to get PWM6/servo6 onto output 2.

Right now this is how they're mapped.

Port on FC                    Output
M1                               ESC/Motor
M2                               Elevator (servo 3)
M3                               Aileron (Servo 4)
M4                               Aileron2 (Servo 5)

Servo 6 is the rudder which obviously is no use here as i don't have another motor/servo output physically.

If I can figure out how to get Servo6 onto M2 (Servo2) then i'll be golden. Obviously aileron2 could go bye bye too but it's 6 of one 1/2 dozen of the other :D

shellixyz, 2017-12-16:
Please clarify I don't understand what you are trying to do.

wx4cb, 2017-12-30:
ok. 

my board(s) as you know only have 4 esc outputs. M1-M2-M3-M4.

Because there is no such thing as resource reassignment in INAV and the fact that i'm only using 1 motor (on M1) that leaves me with just elevator and 1 aileron channel for servos (M3/M4). This means that any aircraft with a rudder cannot be used on these boards because there isn't the available ESC/Servo output to drive them.

This was why I was trying to create a mix where there's only 4 motor outputs needed (as described in the last post). Obviously if they would allow resource reassignment this is a null and void issue. but it would be nice to create a mix where there's only 1 motor and the next 3 "channels" are servos as below:

M1 ESC/Motor
M2 Elevator (servo1 )
M3 Aileron (Servo 2)
M4 Rudder (Servo 3)

shellixyz, 2018-01-15:
You only need the patch to disable the second motor (already provided) and make a custom mix as you said. Resource management would not help other than not needing the patch for disabling the second motor.

wx4cb, 2018-01-15:
@shellixyz thing is. I couldn't find any documentation on how to make a custom mix - unless i'm blind, but like you i've been away with work and the family crap we have to deal with this time of year :D

shellixyz, 2018-01-16:
@wx4cb Here it is ;)
- [Mixer.md](https://github.com/iNavFlight/inav/blob/master/docs/Mixer.md)
- [Video: Setting up custom servo mixes in iNav (or Cleanflight/Betaflight) for V Tail by Painless360](https://www.youtube.com/watch?v=qL2LMFF5bJ4)

DzikuVx, 2018-04-25:
@wx4cb has mixer GUI in Configurator so if you do not mind, I'm gonna close this one

closed 2018-04-25 by DzikuVx