iNavFlight/inav #2596: Unexpected maximum distance and fixed RSSI reading on OSD

fox68k, 2017-12-16:
### Board and Version
Omnibus F4 V2.1 Flight Controller STM32 F405 MCU Integrated OSD Built-in 5V BEC Current Meter
https://www.banggood.com/BF3_1_5-Omnibus-F4-V2-Flight-Controller-STM32-F405-MCU-Integrated-OSD-Built-in-5V-BEC-Current-Meter-p-1143258.html
1.8.0 (latest stable release at the time of writing)

# INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

### Behavior
* The flight summary reports max distance 2.58km when flight footage shows it should be 2.05km. Also, RSSI value is fixed to 99 even though I am pretty sure it should be lower (betaflight seems to be reading well). I also checked at 4.5km from home and still same reading.
Flight footage: https://youtu.be/D9QHK8-JHZg
* No special steps to reproduce. One thing i noticed is i return close to home to make another small trip up to ~570m from home before ending flight at launch point.
* It should state 2.05km maximum distance.
* It displays 2.58km as maximum distance.

# dump

# version
# INAV/OMNIBUSF4PRO 1.8.0 Nov  1 2017 / 06:34:32 (912d1315)

# resources

# mixer
mixer QUADX

mmix reset

# servo
servo 0 1000 2000 1500 100 -1 
servo 1 1000 2000 1500 100 -1 
servo 2 1000 2000 1500 100 -1 
servo 3 1000 2000 1500 100 -1 
servo 4 1000 2000 1500 100 -1 
servo 5 1000 2000 1500 100 -1 
servo 6 1000 2000 1500 100 -1 
servo 7 1000 2000 1500 100 -1 

# servo mix
smix reset

# feature
feature -RX_PPM
feature -VBAT
feature -RX_UIB
feature -RX_SERIAL
feature -MOTOR_STOP
feature -SERVO_TILT
feature -SOFTSERIAL
feature -GPS
feature -TELEMETRY
feature -CURRENT_METER
feature -3D
feature -RX_PARALLEL_PWM
feature -RX_MSP
feature -RSSI_ADC
feature -LED_STRIP
feature -DASHBOARD
feature -BLACKBOX
feature -CHANNEL_FORWARDING
feature -TRANSPONDER
feature -AIRMODE
feature -SUPEREXPO
feature -VTX
feature -RX_SPI
feature -SOFTSPI
feature -PWM_SERVO_DRIVER
feature -PWM_OUTPUT_ENABLE
feature -OSD
feature VBAT
feature GPS
feature RSSI_ADC
[... 490 lines cut ...]
set nav_mc_pos_z_d = 0
set nav_mc_vel_z_p = 70
set nav_mc_vel_z_i = 50
set nav_mc_vel_z_d = 10
set nav_mc_pos_xy_p = 55
set nav_mc_pos_xy_i = 120
set nav_mc_pos_xy_d = 10
set nav_mc_vel_xy_p = 150
set nav_mc_vel_xy_i = 15
set nav_mc_vel_xy_d = 100
set nav_fw_pos_z_p = 50
set nav_fw_pos_z_i = 0
set nav_fw_pos_z_d = 0
set nav_fw_pos_xy_p = 75
set nav_fw_pos_xy_i = 5
set nav_fw_pos_xy_d = 8
set rc_expo = 70
set rc_yaw_expo = 20
set thr_mid = 50
set thr_expo = 0
set roll_rate = 36
set pitch_rate = 36
set yaw_rate = 32
set tpa_rate = 0
set tpa_breakpoint = 1500

shellixyz, 2017-12-17:
For the max distance my bet is on a GPS glitch. For a moment your GPS thought you were 2.58km away even when you really were only 2.05km away.

fox68k, 2017-12-18:
Not sure how this maximum distance is computed but my understanding is it should not be possible to display anything that has not been displayed previously on the OSD. I think this should be looked into it a bit more.

Regarding fixed RSSI reading, I have more details: I am using a L9R with analog RSSI (3.3v).
Additionally, I got the following footage: https://youtu.be/XYGAZxcTNDM which shows the values suddenly jumping from 99 down to 49 (or less) and same back to 99. It does not look good.
Any ideas?

shellixyz, 2017-12-18:
Max distance is updated each time the distance is more than recorded max distance. I don't think there is a bug there it is very simple.

```
if (stats.max_distance < GPS_distanceToHome)
            stats.max_distance = GPS_distanceToHome;
```

Because you didn't see it on the OSD doesn't mean it didn't happen it can only be 1 measurement and I don't know the OSD refresh rate but the GPS refresh rate is probably higher and it possible it hasn't even been displayed at all on the OSD.

shellixyz, 2017-12-18:
I don't know about your RSSI issue I never used analog RSSI. I checked iNav and Betaflight both read the analog RSSI on the same pin but maybe it is computed differently.

fox68k, 2017-12-18:
This is quite possibly the RSSI issue I am experiencing: https://github.com/iNavFlight/inav/issues/2509

shellixyz, 2017-12-18:
Not sure because your rssi_scale is 100 and the issue in #2509 showed up only when this was set to a different value but you can test #2510 anyway it can't hurt.

sonnykaye, 2017-12-18:
Im not sure where to ask this but is stick mixing implemented in some
fashion in inav? like it is in Arducopter? I am working on a project
involving blend of autopilot and teleoperator

shellixyz, 2017-12-18:
@sonnykaye You can try the slack channel or rcgroups but in an issue about RSSI and OSD issues it's clearly not the right place

sonnykaye, 2017-12-18:
@shellixyz thank you for being courteous about it.
I knew it wasn't the right place but got a bit impulsive.
Ill check the slack. Good luck with your issue.

fiam, 2018-02-25:
Managed to reproduce the distance part.

fiam, 2018-02-26:
I'll be closing this one once #2837 is merged. RSSI issue was probably fixed in #2509. If the RSSI problem is still present, please open a new issue.

digitalentity, 2018-02-26:
Fix merged. Closing.

closed 2018-02-26 by digitalentity

padcom, 2018-04-22:
I am experiencing the exact same situation with L9R and Omnibus F4 AIO board. Can someone confirm if #2509 actually fixed that?

PR #2837 (2018-02-26): Fixes for incorrect distance calculations in osdFormatDistanceSymbol()
Fixes #2596 and  #2745
commit a195a7ce (2018-02-26): Fix calculation for kms in osdFormatDistanceStr()

Reported incorrect numbers in maximum and total travelled distances
in the OSD.

Fixes #2596