iNavFlight/inav #2598: Feature Request: in-flight OSD access

csurf, 2017-12-17:
here's a FR for in-flight OSD access, similar to that of the ET vector

- allow access to the OSD (via a toggle switch) while "in flight" / "armed" in order to be able to adjust some of the FC params while in the air. 

- Mainly, allow access to PID's, rates, trim offsets, etc., as well as the level offsets (for auto-level flight). Obviously, don't allow access to anything "dangerous" that might cause a crash, or anything that requires rebooting the board.

- It would probably be preferable (and easier) to setup a slimmed-down 'flight OSD' that's separate from the main configuration OSD that's simple to navigate and holds only those options that can be configured during flight.

- the toggle switch could either work as a simple 'on/off' toggle, or perhaps we could try to setup something that emulates the vector's 'multi-toggle' where a certain switch would need to be toggled multiple times during a short period of time in order to activate the flight OSD.

- The plane/model would need to be placed into a special "OSD cruise mode" that maintains altitude, ground speed, and heading. This would probably mean just forcing it into a combination of ALTHOLD and POSHOLD or CRUISE mode (was CRUISE ever finally implemented for FW?)

- for copters, it could either use POSHOLD or CRUISE mode (user-configurable). IOW, force the copter into either a hover, or maintain it's course/heading if it's already in motion

- some sanity checks would need to be implemented in order to prevent the user from going into OSD mode under dangerous conditions. Some checks that I can think of are 'minimum height' (to avoid going into the OSD too close to ground), 'minimum & maximum ground speed' (for FW, to avoid stalling and/or hazardous high-speed cruising), and GPS fix / HDOP / Sat count (to avoid doing a CRUISE or POSHOLD with a sketchy GPS lock)... Perhaps battery level could also be taken into account, but I'm not sure if this would be very helpful...

I believe that this would greatly improve & speed up the in-flight tuning process to be able to visually change the various flight parameters in a safer, more controlled manner (versus having to blindly fiddle with radio switches while still trying to fly the plane/copter).

(FYI, this is meant mainly for boards with an on-board OSD, but perhaps it would be reasonable to expand this to setups with OSD over serial)

MW2Stambo, 2017-12-17:
You could just display your PIDs on the OSD during tuning flights and use in flight tuning functions already available.
Would be nice to have a couple of different OSD displays available though which I think is planned for 1.9.
See #2477

csurf, 2017-12-18:
yes, that is true, but the point of this feature request is to provided expanded functionality similar to that of the ET Vector, which has a true 'smart' inflight OSD that allows you to easily configure everything via a menu system and stick navigation while the FC maintains stable flight.

bakchos1, 2018-01-18:
If we could start with the option to turn off th OSD in flight, it would be great to have a clean screen when we want to enjoy the view

closed 2018-05-27 by stale[bot]

a1axx, 2020-12-06:
Will this ever be revisited? Inflight menu would be excellent